naturerobots/move_base_flex #11: nav_core2

reinzor, 2017-11-21:
Are you planning to be compatible with https://github.com/locusrobotics/robot_navigation/tree/master/nav_core2 ?

corot, 2017-11-22:
Not by now, but we support nav_core-based pluggins, so I suppose we can load too nav_core2 plugins using the https://github.com/locusrobotics/robot_navigation/tree/master/nav_core_adapter package.

reinzor, 2017-11-27:
> locusrobotics/robot_navigation/tree/master/nav_core_adapter package.

Thanks

-Rein

reinzor, 2017-11-28:
Thanks for your answer @corot. I have been looking at the abstraction you have created for the controller and the planner (https://github.com/magazino/move_base_flex/blob/master/move_base_flex_core/include/move_base_flex_core/abstract_local_planner.h https://github.com/magazino/move_base_flex/blob/master/move_base_flex_core/include/move_base_flex_core/abstract_global_planner.h). I also looked at the ones defined here in the nav_core2 interface (https://github.com/locusrobotics/robot_navigation/blob/master/nav_core2/include/nav_core2/global_planner.h https://github.com/locusrobotics/robot_navigation/blob/master/nav_core2/include/nav_core2/local_planner.h). 

I think the interfaces you defined are nice, except for the interfaces in the local planner for computing the velocity and the isGoalReached method. Because the state used in this method is not an input argument but magically obtained via the TF listener or an odom subscriber. In the nav_core2 interface, this has been changed. What do you think about this?

corot, 2017-11-28:
I agree; #17

reinzor, 2017-11-28:
Okay, I will close this issue

closed 2017-11-28 by reinzor