naturerobots/move_base_flex #13: Allow loading multiple planners and controllers, electable at runtime

corot, 2017-11-27:
Our ROS API already provides a xxx_name field for that, but it's not implemented in the server

reinzor, 2017-11-28:
I think also a service should be exposed to query the available planners that are configured on startup. This way, the configuration of the planners can be done on startup with use of the parameter server.

corot, 2017-11-28:
Well, the list of loaded plugins is already available as a parameter, and it can be changed at runtime through dynamic reconfiguration.

reinzor, 2017-11-28:
How would you like to deal with planner specific configurations?

corot, 2017-11-28:
Each planner has his own namespace on the parameter server, so you can configure each one independently both at launch time and with dyn. reconf. if the plugin offers it

spuetz, 2018-01-02:
I'm on it!

commit 8b019621 (2018-01-28): Handle the no switch necessary case. This closes #13
closed 2018-01-28 by spuetz