naturerobots/move_base_flex #14: Request: Separate the abstraction from the implementation

reinzor, 2017-11-28:
I am browsing through your code (since there is not that much documentation yet, except for your roscon presentation and the tutorial). And what is not clear for me why you have a separate package for the interface definition (move_base_flex_core pkg) and another package (move_base_flex pkg) that holds an abstraction for the executioners and an implementation. Why not totally separate the abstraction from the implementation and create, for example, a mbf_move_base and mbf_simple package? Correct me if this remark does not make sense.

corot, 2017-11-28:
This is gonna happen very soon; take a look at [advanced_naming branch](https://github.com/magazino/move_base_flex/tree/advanced_naming), about to be merged

reinzor, 2017-11-28:
Nice, this is exactly what I meant! Keep up the good work! I am still investigating how your framework functions and how we can apply it in our use case. Thanks again for your quick reply!

reinzor, 2017-11-28:
I was looking at the https://github.com/magazino/move_base_flex/tree/advanced_naming branch and I see two different abstract classes in the core. Also one containing costmap2d specific information, why is this so called MoveBaseController class definition not named Costmap2DControll and placed in the mbf_costmap_2d_nav pkg?

corot, 2017-11-28:
Good question; we are thinking on a nicer way. The two possibility we contemplate are:
- create a separated mbf_xxx_core per navigation server, like mbf_abstract_core, mbf_costmap_2d_core, mbf_simple_core... Too verbose for my taste, but clean and tidy.
- create a separated directory within nav_core, like
  - include/mbf_core/abstract
  - include/mbf_core/costmap
  - include/mbf_core/simple
  - ...   

  more compact, but maybe less clear.

What you propose, move the interfaces to mbf_costmap_2d_nav has the problem of making your plugins depend on a relatively heavyweight package as mbf_costmap_2d_nav, instead of on a tiny, headers-only package like mbf_core, or nav_core, or nav2_core.

What would be your choice?

reinzor, 2017-11-28:
I should take into account the following considerations:

- The planners depend on the planning representation and thus the server, e.g. mesh, gridmap, costmap 
- You don't want to be dependent on a representation that is not for your planner (this excludes the second option)

So option one is indeed verbose but IMO the most neat solution, or just adding the interface in the representation specific package (e.g. mbf_costmap_2d_nav) since you want to use the representation in your plugin anyway.

corot, 2017-12-11:
That's indeed what we have implemented on advanced_naming branch (to be merged soon)

closed 2017-12-11 by corot