naturerobots/move_base_flex #16: Shouldn't the monolithic move_base action implemented on mbf_costmap_2d_nav?

corot, 2017-11-28:
Now is on mbf_abstract_nav

reinzor, 2017-11-28:
If this state machine code is going to be moved, I would suggest moving it to another node to keep the mbf_costmap_2d_nav server clean:

- mbf_costmap_2d_nav server (exposing the https://github.com/magazino/move_base_flex/tree/advanced_naming/mbf_msgs interfaces except for the MoveBaseAction)
- mbf_costmap_2d_move_base_node (containing the state machine logic and merge this with the python node functionality present in https://github.com/magazino/move_base_flex/blob/advanced_naming/mbf_costmap_2d_nav/scripts/move_base_legacy_relay.py)

This way, the MoveBase action can be deprecated: https://github.com/magazino/move_base_flex/blob/advanced_naming/mbf_msgs/action/MoveBase.action

corot, 2017-11-28:
mmm....  :thinking:  I don't want to deprecate mbf_msgs/MoveBase action! The first idea behind MBF is to allow external executives to manage navigation, but we also want to overcome the limitations of current move_base, by providing a much more detailed MoveBase action.

A different thing is the move_base_msgs/MoveBase action provided by the move_base_legacy_relay.py scripts, that it's there to allow us to claim that MBF is fully backward-compatible with current move_base, and yes, is deprecated (though not clearly stated... will fix that in the wiki)

reinzor, 2017-11-28:
Ok. If I understand this correctly: you want to provide one action to move the base right? Why not make a separate node then (or even package) that exposes this functionality and calls the other actions of the server? This way, this node can be used for all servers and you make a clear separation between the server and the statemachine logic. 

Basically this is transferring the statemachine logic (that is now in the acstract server), to a separate node.

corot, 2017-11-28:
Very good idea!

corot, 2017-12-22:
:thinking: 
I'm rethinking this a bit... why move_base action cannot be on the abstract server? Once renamed to something with less connotations as Navigate.action, or MoveTo.action, it makes perfect sense to provide this action server as part of any navigation framework, implemented by just calling the other actions.

spuetz, 2018-07-03:
I've divided the actions into separated classes with in the new structure I implemented for the mbf_abstract_nav, which goes along with the plugin concurrency issue #36. So #36 will solve this issue too, more or less.

corot, 2018-08-24:
Right, this actions get implemented as a MoveBaseAction object, same way as Controller/Planner/RecoveryAction are. Not sure if it's the cleanest option... but makes sense to me, and so I'll let it as it is by now. @reinzor suggestion of a external node makes sense too, though, so I'll keep the idea in mind

closed 2018-08-24 by corot