naturerobots/move_base_flex #19: Best practices for using MBF

james-ward, 2017-12-20:
With the latest changes to master I am changing my code to suit the new way of doing things.
I have a question about what the intended way to use this is.

Should I be creating my classes to inherit directly from the various Abstract* classes, or are the Simple* classes designed to be subclassed?
#14 seems to imply that everything should directly inherit from Abstract*, but I'm not sure.

corot, 2017-12-20:
What are you trying to do? Create your own MBF-based navigation server? Then you extend Abstract* classes. Simple* classes are the simplest possible example of a navigation server using MBF.

Anyway, sorry for the lack (still) of extensive documentation and tutorial. Will get better soon :grimacing:

james-ward, 2017-12-20:
Thanks for the quick response.
Yes, the aim is to create our own MBF based nav server.

So the Simple* classes are really a demonstration of how to do it? Not really designed to be used as a basis for more complicated classes. (I notice that the CostMap2D implementation inherits directly from Abstract*).

corot, 2017-12-20:
Right, Simple* classes are a use case. We will rename soon, dropping the "Simple", as it looks somehow misguided. You make your server extending from Abstract*, as Costmap2D* and Simple*.

May I ask, what's the particular navigation "flavor" you aim for?

corot, 2017-12-20:
I consider this as answered; please re-open if not!

closed 2017-12-20 by corot