naturerobots/move_base_flex #20: Robot pose not correctly sent to controller

james-ward, 2017-12-20:
The call to `getRobotPose()` doesn't seem to be working.
My controller gets passed an uninitialised `StampedPose` - everything is zero and the strings are all blank.

james-ward, 2017-12-21:
I found the TODO in `AbstractControllerExecution` saying this isn't implemented yet.

spuetz, 2018-01-02:
Do you mean while using 'computeVelocityCmd'? So in the following method in AbstractControllerExecution? If so, you're right! That feature is coming in the next days. 
`uint32_t AbstractControllerExecution::computeVelocityCmd(geometry_msgs::TwistStamped &vel_cmd,
                                                                           std::string& message)
  {
    // TODO calculate robot pose and velocity
    geometry_msgs::PoseStamped robot_pose;
    geometry_msgs::TwistStamped robot_velocity;
    return controller_->computeVelocityCommands(robot_pose, robot_velocity, vel_cmd, message);
}`

spuetz, 2018-01-28:
The robot pose is now computed by the ControllerExecution and given to the plugin. The velocity is still missing. I'll open another issue for the robot velocity.

commit fa63625a (2018-01-28): Compute robot pose for controller plugin, this closes #20
closed 2018-01-28 by spuetz