naturerobots/move_base_flex #30: issue compiling

henoSH, 2018-01-10:
Hi I' m receiving many errors while compiling the workspace. Are you sure that this version has no error?

corot, 2018-01-10:
Well, I compile all without problems. Can you paste here the errors you have?

henoSH, 2018-01-18:
here is the error:

/home/ros/catkin_ws/src/move_base_flex-master/mbf_costmap_nav/src/nav_core_wrapper/wrapper_global_planner.cpp: In member function ‘virtual uint32_t mbf_nav_core_wrapper::WrapperGlobalPlanner::makePlan(const PoseStamped&, const PoseStamped&, double, std::vector<geometry_msgs::PoseStamped_<std::allocator<void> > >&, double&, std::string&)’:
/home/ros/catkin_ws/src/move_base_flex-master/mbf_costmap_nav/src/nav_core_wrapper/wrapper_global_planner.cpp:53:68: error: no matching function for call to ‘nav_core::BaseGlobalPlanner::makePlan(const PoseStamped&, const PoseStamped&, std::vector<geometry_msgs::PoseStamped_<std::allocator<void> > >&, double&)’
   bool success = nav_core_plugin_->makePlan(start, goal, plan, cost);
                                                                    ^
/home/ros/catkin_ws/src/move_base_flex-master/mbf_costmap_nav/src/nav_core_wrapper/wrapper_global_planner.cpp:53:68: note: candidate is:
In file included from /home/ros/catkin_ws/src/move_base_flex-master/mbf_costmap_nav/include/nav_core_wrapper/wrapper_global_planner.h:44:0,
                 from /home/ros/catkin_ws/src/move_base_flex-master/mbf_costmap_nav/src/nav_core_wrapper/wrapper_global_planner.cpp:41:
/home/ros/catkin_ws/src/nav_core/include/nav_core/base_global_planner.h:57:20: note: virtual bool nav_core::BaseGlobalPlanner::makePlan(const PoseStamped&, const PoseStamped&, std::vector<geometry_msgs::PoseStamped_<std::allocator<void> > >&)
       virtual bool makePlan(const geometry_msgs::PoseStamped& start, 
                    ^
/home/ros/catkin_ws/src/nav_core/include/nav_core/base_global_planner.h:57:20: note:   candidate expects 3 arguments, 4 provided
make[2]: *** [move_base_flex-master/mbf_costmap_nav/CMakeFiles/mbf_nav_core_wrapper.dir/src/nav_core_wrapper/wrapper_global_planner.cpp.o] Error 1
make[1]: *** [move_base_flex-master/mbf_costmap_nav/CMakeFiles/mbf_nav_core_wrapper.dir/all] Error 2
make: *** [all] Error 2

corot, 2018-01-18:
mmm... :thinking:  are you still using indigo?

henoSH, 2018-01-18:
yes

corot, 2018-01-18:
We didn't target indigo from the beginning, sorry. I should have written that somewhere.

But making it compile will be trivial. Just remove the "cost" parameter on makePlan call.

henoSH, 2018-01-19:
yes there is no information with which distribution this package can be integrated!

corot, 2018-01-23:
Right. Added to the README. Thanks!

closed 2018-01-23 by corot