naturerobots/move_base_flex #31: Setting new plan while moving warning in Turtlebot navigation

SHTseng, 2018-01-25:
Hi, 
I would like to thank you for developing such great project. I am currently working on integrating this project with my research project which uses Turtlebot2. However, I've encountered some problems that need some helps.

I am using Ubuntu 16.04 with Kinetic. I set up the navigation environment by launching the nodes mbf_costmap_nav and move_base_legacy_relay in my launch file.

1. I publish the goal via rviz, the robot can't not finish the execution of the path and stopped randomly. Also I have tried using both dwa_local_planner and basic trajectory_planner. The error code I got is:

[ WARN] [1516878140.884170342, 123.947000000]: Setting new plan while moving!
[ERROR] [1516878140.884297843, 123.947000000]: Could not start moving, because another moving thread is already / still running! Canceling the action call.
[ WARN] [1516878140.884486489, 123.947000000]: Recovery behaviors are disabled!
[ WARN] [1516878140.884515078, 123.947000000]: Abort the execution of the controller: 

I think the errors come from functions `setNewPlan` and `startMoving` in the class of `mbf_abstract_nav::AbstractControllerExecution`. 
Do you guys have any suggestions on how to fix this problem?

2. Many Turtlebot navigation parameters are scripting in the yaml format, I want to load these parameters to the MBF node. Is it possible to implement another script (similar to move_base_legacy_relay) to porting these parameter to MBF?

Thanks again!

SHTseng, 2018-01-25:
After tracing the code, I think the problem is the planner get new goal in every planner frequency. The new path is sent to the path execution action server while the robot is still moving (moving_ == true). Thus the `startMoving` in `AbstractControllerExecution` returns false.

Currently, I disabled the `plannerThread` by commenting the `planner_cond` variable in the `callActionMoveBase`. After that everything return to the same as move_base.

corot, 2018-01-25:
Hi! I already detected the problem when continuously re-planning and using the separated servers (get path, exe path...).  But didn't experience with the monolithic move_base. I think I fixed it on branch issue_31.

SHTseng, 2018-01-25:
Thanks for the reply, I will check the branch issue_31!

corot, 2018-01-25:
So yes, makes plenty of sense that you avoid the problem by disabling plannerThread. But can you please try the new branch? Should also work without stopping continuous replanning

corot, 2018-01-25:
For your second question...  not sure what the problem is. Normally you should have your move_base_flex.yaml parameters file, following the new naming. But,,, the legacy scripts loads his own parameters file with the old naming, and it tries to reconfigure move_base_flex with them, using dynamic reconfigure.

SHTseng, 2018-01-26:
I have tested dwa_local_planner and teb_local_planner with continuous replanning, both of them work really well, and I think I know how to solve my second question. Thank you so much for solving this issue so quickly.

Also, I have another question. To my understanding, the design of legacy move_base does not calling the global planner in planner frequency, it only accept preempt requests? So, is this a new feature for move_base_flex which different from legacy move_base? Please correct me if there is anything wrong, Thanks!

corot, 2018-01-26:
Not sure if I understand your question... but the point is that MBF offers separated action servers for planning, controlling and recovering, so you can decide when to call each from an external executive.
But it also offers a "monolithic" move_base action like the one in the old move_base but with much richer interface. This action is internally implemented by calling to the other 3 action servers.
And finally you have the old move_base action provided for backward compatibility: it works by just calling the new move_base action server.

corot, 2018-01-26:
Merged into master

closed 2018-01-26 by corot