naturerobots/move_base_flex #35: Support for AckermannDrive messages

james-ward, 2018-02-01:
At the moment mbf generates a `TwistStamped` message like the previous nav stack. This isn't great for non-holonomic robots.
Is it possible to change the type of command message returned to something like `AckermannDrive`?

I'm happy to help with the implementation - are there any suggestions as to the best way to do this given the way you have architected the rest of the project?

spuetz, 2018-02-03:
Hi James,

we thought about the Ackermann messages to integrate. An integration will not happen before the release. But for the next version you could also contribute, that would make the integration much faster. First we should then discuss the integration here, since we need another base class for the plugins, at least the controller plugin. 

However, I think supporting the Ackermann message type and therefore a better support of a car like navigation is a good idea and extension!

james-ward, 2018-02-03:
I'm happy to do it. It seems to me that it would require a pretty fundamental change to the entire stack - the idea of a velocity command being a TwistStamped is pretty embedded into the stack. We'd need to change the base controller class, but also the base controller execution class, and the server I think.
Is this the best way to do it? How would we support two hierarchies of classes where the main difference is really in the type of one of the arguments to some of the functions?

james-ward, 2018-02-04:
It seems to me that this could be done with templates, using TwistStamped as the default. The drawback is that it makes things look a bit messier with the repeated use of <> everywhere. What do you think?

spuetz, 2018-02-09:
Templates are determined by compile time, so we would need different classes here for the navigation server which would mean an additional overhead. Maybe another way would be better. Simply having another base class like: mbf_core::abstract_ackermann_{planner, controller, recovery} should be fine. The classes abstract_ackermann_* and abstract_* could both inherit {planner, controller, recovery}_base. And then we simply try to load both and determine the type by successfully loading one or the other. Better ideas?

james-ward, 2018-02-12:
This approach is good. It will also require doing the same thing for (at least) abstract_controller_execution because this also assumes a TwistStamped message.

spuetz, 2018-02-15:
I think we will implement the support for Ackermann msgs for the next release. How do you use mbf in the current state?

james-ward, 2018-02-15:
That's great news.
Currently we use our own controller that returns a TwistStamped message, but only with the linear.x and angular.z non-zero. This can then be turned into a steering angle and speed.
However, one big problem is that if you command a linear.x speed of 0, there is no way to resolve the steering angle. So coming to a stop and still maintaining steering is not trivial.