naturerobots/move_base_flex #37: AbstactMap interfaces?

naveedhd, 2018-02-03:
Since the move_base_flex allows to make implementations using different map representations (costmaps/gridmaps etc..), has there been a thought of making something like a AbstractMap from which one of these could be inherited/plugin-ed?

To give a context, I toyed around gridmaps and made `mbf_gridmap_core` and `mbf_gridmap_nav` which are just plain translations of `mbf_costmap` for now replacing `costmap2dros` with `customgridmapros`. If somehow the the map representation could be abstract and inherited or loaded as a plugin, one navigation server can be used with different map representations. Examples of features that we might get are a) use components A, B, C with costmap, and D, E, F with gridmaps b) use planner A with gridmap and controller B with gridmap.

If we have two set of components(planner/controller/recoveries) , one working with costmaps and other with gridmaps, two separate instances of navigation servers have to be launched.  

Although this idea seems to me problematic in many ways (eg. api is different for each etc) but would like to know thoughts on this.

Thanks for the package 👍

spuetz, 2018-02-09:
Hi @naveedhd, sry for the late response. We still investigate in the `mbf_gridmap_nav` and `mbf_gridmap_core` package. 

Interesting thoughts on how to fuse all navigation servers to one and loading the map representations as plugins. As you said its complicated to combine all. So I need more time to think about that.

corot, 2018-03-12:
Hi @naveedhd, did you make public your mbf_gridmap_core and mbf_gridmap_nav packages?
I would like to try them and possibly integrate on MBF.
Thanks!

naveedhd, 2018-03-12:
Hi @corot, No.. we are kind of blocked/indecisive with our requirement of using existing and new planners/controller under one roof (one nav server i.e. costmap or gridmap). We are proceeding for now with costmap because of this constraint.

I just have half touched `mbf_gridmap_core` and `mbf_gridmap_nav` (just replica of costmap counterparts) and `gridmap_ros` (working with aggressive marking and clearing of laser scans) and no current ongoing or near future work aligned.

One of the question I encountered with gridmap is the requirement of duality i.e. `global` and `local` and it seemed possible to just get away with one for both. Want to continue working on this but lacking a sense of direction and time.

naveedhd, 2018-03-12:
Please share if there are any plans and suggestions for mbf_gridmap or mbf in general close to this.

Also please close this issue if you want.

jorgemia, 2020-11-13:
Are there any implementations anywhere of grid_map with mbf?

spuetz, 2020-11-13:
Not yet public, will be end of January.

jorgemiar, 2021-03-08:
Hi @spuetz has the grid_map implementation been released anywhere?

ingramator, 2021-03-10:
@spuetz  did you end up releasing this grid_map compatibility? I know there was work with the original move_base so would be great to see with mbf!

spuetz, 2021-03-10:
Hello, glad there is interest in it. We are still in development and have not released it yet. But that is definitely still the plan to make this available in the next few months. Sorry that I have to put you off until then. I guess that we can expect a first working version at the end of May.

ju-mingyue, 2021-11-22:
@Rayman  When I use the move_base_flex function package again, I follow the link [https://github.com/uos/ceres_robot/blob/master/ceres_navigation/launch/move_base_flex.launch](url) to configure the file, but after running it, it is found in the node graph that this function package does not subscribe to the map and the lidar signal (the topic here is corresponding), and I can’t read it To the planned path.I don’t quite understand what does move_base_legacy_relay do?

soldierofhell, 2022-04-08:
Hi @spuetz, any chance to publish grid_map integration? It would be a good starting point for nav2 development

shastro, 2023-05-18:
I am also very interested in this, we have been using GridMap on our robot because of its nice representation and filtering options. Does anyone know any way to plan using a gridmap? We have been trying to convert to `costmap_2d` and use with move_base or mbf_costmap2d, but it has been hard to find resources on how to do this. It seems there are not many planners that use GridMap yet even after all this time.