naturerobots/move_base_flex #58: Force config. regeneration when adding new params

corot, 2018-03-19:
When adding new params to https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/src/mbf_abstract_nav/__init__.py, you need to touch https://github.com/magazino/move_base_flex/blob/master/mbf_costmap_nav/cfg/MoveBaseFlex.cfg before compiling so code for the new parameters is generated.

Cannot do better?

corot, 2019-06-16:
One ugly option is to always touch the .cfg file, but this makes compilation longer when only it's occasionally needed:

- file(TOUCH_NOCREATE cfg/MoveBaseFlex.cfg)    requires CMake 3.12
or
- execute_process(COMMAND touch cfg/MoveBaseFlex.cfg)

jspricke, 2019-06-24:
You would need to add the file as a dependency to the add_custom_command in the generate_dynamic_reconfigure_options macro, i.e. here:
https://github.com/ros/dynamic_reconfigure/blob/melodic-devel/cmake/dynamic_reconfigure-macros.cmake#L73
Maybe an extra parameter in for the macro works.
I didn't find a way to inject the dependency later.

corot, 2019-06-25:
> Maybe an extra parameter in for the macro works.

Unfortunately not; generate_dynamic_reconfigure_options tries to process it as any other .cfg file

jspricke, 2019-06-25:
>> Maybe an extra parameter in for the macro works.
>
>Unfortunately not; generate_dynamic_reconfigure_options tries to process it as any other .cfg file

I meant you can improve the generate_dynamic_reconfigure_options (i.e. 
patch it), such that it supports your use case.

corot, 2019-06-25:
> I meant you can improve the generate_dynamic_reconfigure_options (i.e. patch it)

:fearful:  I know the absolute minimum about CMake...