naturerobots/move_base_flex #65: Switching Controller/Planner

predoli, 2018-04-11:
Hey, great work with move_base flex!
I want to implement a additional planner layer that runs on execution time.
So maybe one idea was to implement this as local planner and then call them in a sequence. Is this already possible in move_base_flex?

spuetz, 2018-04-11:
You are welcome. Can you explain your idea in detail. I don't get it. Thank you.

predoli, 2018-04-11:
Thanks to the fast response @spuetz 
Sorry for my bad declaration.
Basically i want a behavioral layer between local and global planner that runs in real time and modifies the global plan and adding a velocity profile to it (by using the time stamps). 
My local planner already uses them.
My initial plan was to write a own plug-in type and crush that into the standard move base. 
But since I saw this package i thought i could avoid that, by defining my behavioral layer as a controller and switch controller after in every time step. Like Plan path -> Execute control using behav. planner -> switch controller to local planner -> execute control using standard local planner -> switch back to behav. layer planner and so on.

corot, 2018-04-29:
Hi @predoli! So... If I understand well, wouldn't be enough if the local planner planner can follow a continuous stream of global plans, and switch immediately to the latest one?
If not, I'm afraid I need more explanation to understand your idea

spuetz, 2018-07-03:
@predoli Would it make sens to run these two controllers simultaneously?

- execute control using behav. planner
- execute control using standard local planner

predoli, 2018-07-03:
Hey @spuetz @corot  thanks for  your responds. 
As of now i just broke the standard move base and implement everything proprietary.

> Would it make sens to run these two controllers simultaneously? Like using #36 ?

It would totally make sens to do this, but they must share some memory. Since the standard local planners input path (normally the global path) in my case, is the result of this behav. planner.

Oli

corot, 2018-08-24:
Hello bacj @predoli; we have merged the concurrency feature: now you can run N planners and M controllers in parallel. Not sure if this will help you with your architecture. If not, feel free to close this issue.

Maybe would make sense to propose your usecase for the ongoing discussion about ROS2 navigation

spuetz, 2018-10-08:
Hey @predoli, you could run your problem now: If your `behav_planner` is calling the `ExePath` action with `controller = local_planner` on another `concurrency_slot` after it modified the plan it got. And also your `local_planner` needs to call the `ExePath` with `controller = behav_planner` on the original  `concurrency_slot`.

spuetz, 2018-11-12:
@predoli Any news?

predoli, 2018-11-13:
@spuetz yes, i have a small example running, related to the tutorial with SMACH. Thanks a lot for your work.

closed 2018-11-13 by predoli