naturerobots/move_base_flex #66: Question about computeVelCmd inheritance in mbf_costmap_nav

SHTseng, 2018-04-20:
Function computeVelocityCmd in costmap_controller_execution intends to override the one in base class. 
[computeVelocityCmd in abstract](https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/include/mbf_abstract_nav/abstract_controller_execution.h#L207),     [computeVelocityCmd in costmap_nav](https://github.com/magazino/move_base_flex/blob/master/mbf_costmap_nav/include/mbf_costmap_nav/costmap_controller_execution.h#L85) 

To my understanding, the override function needs to take same inputs as base class function in c++. I've test it with output message which shows the move_base_server is calling the abstract one. Also, I'm using the latest commit in master branch.

Just want to clarify if this is an intentional move, please correct me if there is any misunderstanding, thanks!

corot, 2018-04-29:
You are right. Thanks for detecting it! I PRequested a patch

corot, 2018-05-10:
Hi @SHTseng. I merged a fix. Would you mind to give it a second try to master branch?
Thanks!

SHTseng, 2018-05-15:
Hi, I've tested it with turtlebot, now it is calling the overriding function and works well. 
Thanks for the fix!

closed 2018-05-15 by corot