naturerobots/move_base_flex #70: Better handling continuous replanning

corot, 2018-04-29:
As we use now simple action server for exe_path, every new plan will stop the action, set the new plan, and restart exe_path. Following #64, this is not only ugly, but also will introduce a zero-velocity command at every new plan.

Possible solution: use a normal action server so every new goal received while running a previous one just updates the path we are already following, without preempting the previous goal

corot, 2018-08-24:
The new concurrency feature opens new options to implement this issue.
By other hand, now I get #79

corot, 2018-09-04:
Actually, the new concurrency feature makes implementing this harder, at least at first glance, as we treat all actions homogeneously (what is quite elegant) and adding an "exception" for the controller when already running will look very hackish. More precisely, when already navigating the controller receives a new goal, I want to avoid [this cancel](https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/include/mbf_abstract_nav/abstract_action.h#L43) and call instead setNewPlan. But for that I need to entirely rewrite the AbstractAction::start method.
 
Unless anyone has a better idea (hence the _"help wanted"_ tag, I propose to add a ChangePath (or SetNewPath) service that bypass entirely the action and changes the path of currently running ExePath action (failing if none)

spuetz, 2018-09-05:
- An idea is to have a controller and a planner sharing a plan object. The controller plan is adapted while the planner is planning more and more of the path.
- Or we could invent a service for setNewPlan.
- Check if a controller with the name and the slot number is already running and then just pass the plan to setNewPlan.

spuetz, 2018-10-08:
@corot Do you think we need to investigate more within this feature?