naturerobots/move_base_flex #72: move_base_flex is not actually a metapackage

gavanderhoorn, 2018-06-02:
This is more a question than an actionable item, but I was just wondering why the description of `move_base_flex` claims it to be a metapackage, while it isn't actually one.

The name *metapackage* has an 'official' definition and defined semantics ([REP-149](https://github.com/ros-infrastructure/rep/blob/d9d296225e86b490e922e3f253385c6f00c184bf/rep-0149.rst#metapackage)), and `move_base_flex` doesn't seem to adhere to that.

Is `move_base_flex` actually more a convenience package that users can `find_package(..)` to get the entire set of MBF dependencies / often used pkgs, instead of having to list all dependencies separately?

corot, 2018-06-03:
Good question...  After reading the REP, makes sense to me marking it as metapage.
@spuetz , what do u think?

gavanderhoorn, 2018-06-03:
Well, my comment was actually more: it claims it's a metapackage, but it isn't at the moment.

corot, 2018-09-10:
What is missing is to export the metapackage tag, right?
```
  <export>
    <metapackage/>
  </export>
```
I'll add it. Immediate benefit is that I can compile all MBF packages in one command without compiling the whole workspace

gavanderhoorn, 2018-09-10:
Please check [REP-149](https://github.com/ros-infrastructure/rep/blob/d9d296225e86b490e922e3f253385c6f00c184bf/rep-0149.rst#metapackage). The current [CMakeLists.txt](https://github.com/magazino/move_base_flex/blob/1354252dbf22c672f3e0ba15f0e169672029ac23/move_base_flex/CMakeLists.txt) also contains "illegal" (or at least: out of spec) content for a metapackage.

spuetz, 2018-09-10:
@gavanderhoorn and @mgruhler Thank you guys!

commit c800bd8c (2018-09-10): fix metapackage, closes #72
closed 2018-09-10 by spuetz