naturerobots/move_base_flex #75: ROS2 Plans?

mkhansen-intel, 2018-06-29:
Are there any plans to port move_base_flex to ROS2? I'm driving an effort to create a ROS2 Navigation system, and many of the improvements in move_base_flex are similar to the ideas & concepts we've been discussing within our team. I'd like to make sure we're not working on a redundant effort.

spuetz, 2018-06-29:
Yes! We though about Move Base Flex as a good base for the ROS2 Navigation system. Should we discuss a possible procedure? At the moment I put a lot of work into making MBF parallel. (https://github.com/magazino/move_base_flex/issues/36) This will be released soon. After that I'll have a bit capacities for porting MBF to ROS2.

reinzor, 2019-06-04:
Any updates on this? Will move_base_flex be deprecated in favor of the ROS2 Navigation stack?

spuetz, 2019-06-04:
We planned a MBF version for ROS2, It will be developed when all necessary parts in ROS2 are available.

reinzor, 2019-06-04:
Nice. I am not fully aware of the ROS2 navigation stack features but wouldn't be an idea to merge both efforts?

adamkrawczyk, 2020-04-29:
Any news now?

spuetz, 2020-04-29:
I'll start the MBF Development for ROS2 around September / October.

mkhansenbot, 2020-05-14:
I'm closing this issue as the Nav2 project has been released for over a year now, this question is outdated.

StefanFabian, 2021-04-30:
The existence of Nav2 doesn't have anything to do with a possible ROS 2 release unless the maintainers here stopped their plans in favor of Nav2.
Nav2 is also not perfect and still heavily relies on costmap2d.
Could you or one of the maintainers please reopen this issue as it may still be of interest to other people?

Rayman, 2021-04-30:
The good thing about nav2 is that they have added action interfaces for GetPath and ExePath, just like MBF did for ROS1. For us that was the biggest feature to use MBF.

For ROS2 I think we are going to first try out nav2 as that seems to fulfill all our requirements.

moooeeeep, 2022-03-31:
No pressure, but... any updates on this? :)

Related discussions regarding other map representations at navigation2:

* https://github.com/ros-planning/navigation2/issues/1278
* https://github.com/ros-planning/navigation2/issues/1461

Cakem1x, 2024-10-16:
We just added a [humble](https://github.com/naturerobots/move_base_flex/tree/humble) branch here. It contains our work on migrating move base flex to ROS2. - Check it out!
It might work with other ROS2 versions as well, but ROS2 `humble` is our current target.
As of now, we do not publish precompiled packages, you need to compile it from source.

closed 2024-10-16 by Cakem1x