naturerobots/move_base_flex #93: SIGSEGV crash do to thread unsafe cancel & restart action

howardcochran, 2018-10-04:
I tracked down a memory corruption bug as being caused by lack of thread safety as noted by this comment in mbf_abstract_nav/.../abstract_action.h:
`if(slot != concurrency_slots_.left.end()) // if there is a plugin running on the same slot, cancel it // TODO make thread safe`

External code was sending a RecoveryAction goal, canceling it, then sending another goal the same behavior name and concurrency slot.  The problem occurs if the recovery plugin takes some time to cancel. Since AbstractAction::start() for the canceled action to finish, it would invoke the same recovery behavior object concurrently even though they are in the same concurrency slot. The two threads operating on the same plugin object trashing each other's memory.

howardcochran, 2018-10-04:
### Here is a quick test program that exhibits the behavior when run on a recovery behavior that is non-cancelable or which takes more than a fraction of a second to exit when canceled:

    #!/usr/bin/env python
    
    import sys
    import threading
    import rospy
    import actionlib
    from std_msgs.msg import Header
    import mbf_msgs.msg as mbf_msgs
    
    class RecoverySpammer(object):
        def __init__(self, recovery_name):
            self.recovery_ac = actionlib.SimpleActionClient(
                    '/move_base/recovery', mbf_msgs.RecoveryAction)
            self.recovery_name = recovery_name
            self._is_recovery_running = False
            self._ac_lock = threading.Lock()
            self.timer = rospy.Timer(rospy.Duration(0.5), self._timer_cb)
    
        def start_recovery(self):
            with self._ac_lock:
                rospy.loginfo("Start Recovery")
                goal = mbf_msgs.RecoveryGoal(behavior=self.recovery_name)
                self.recovery_ac.stop_tracking_goal()  # In case old goal exists
                self.recovery_ac.send_goal(goal, done_cb=self._recovery_done_cb)
                self._is_recovery_running = True
    
        def cancel_recovery(self):
            # Note: Not holding self.ac_lock is intentional, as it could deadlock
            rospy.loginfo("Cancel Recovery")
            self.recovery_ac.cancel_goal()
            ## self.recovery_ac.cancel_all_goals()
            with self._ac_lock:
                # stop_tracking: in case cancel tarries for some reason
                self.recovery_ac.stop_tracking_goal()
                self._is_recovery_running = False
    
        def _recovery_done_cb(self, status, result):
            with self._ac_lock:
                rospy.loginfo("Recovery done cb")
                self._is_recovery_running = False
    
        def _timer_cb(self, event):
            if self._is_recovery_running:
                self.cancel_recovery()
                rospy.sleep(rospy.Duration(0.1))
            self.start_recovery()
    
    if __name__ == '__main__':
        rospy.init_node('recovery_spammer')
        recovery_name = sys.argv[1]
        rs = RecoverySpammer(recovery_name)
        rospy.spin()

howardcochran, 2018-10-04:
I have created PR #94 as a request for comments. I do not believe that it is a complete fix for the reasons given in the commit comments. I am intending to start discussion about how to fix this more correctly before I write too much more code.

Here are my assumptions:
1. If a new goal is received for the same concurrency slot that is already running a goal, the existing goal should be canceled (the code actually already did this part).

2. The plugin must not be invoked for the new goal until the previous invocation has finished, even if it takes awhile for it to cancel.

3. If the new goal is requested to be canceled while it is thus waiting, it should be canceled without ever invoking the action plugin

4. The ActionServer's goal_cb function must not block

Here is how I am thinking of changing the existing proposal:
Make AbstractAction::start() just put the new goal on a pending list and start a worker thread for the given concurrency slot if there is not one already running.  Thus, it never blocks.

Change AbstractAction::runAndCleanUp() to be a worker thread loop for a given concurrency slot.  It will remove a goal from the pending list, put it on a "running" list and run the plugin in a loop until the pending list for this slot is empty.

AbstractAction::cancel() will check if the goal ID is on the "running" list. If so, it will call the plugin's cancel method.  If not, check to see if this goal is in the pending list.  If so, remove it.

In order to prevent long queuing of goals, AbstractAction::start() will cancel any goal in the "running" list for this concurrency slot.  If the "pending" list already contains a goal for this slot, replace it, marking the existing one as canceled.

Thus you can see, that, for a given slot, there will never be more than one "pending" and one "running" goal.

Make changes to ControllerAction to adapt it to the changes in data structure without changing its observable behavior.

I am thinking that once a worker thread for a given concurrency slot is started, it will never exit.  Since the slot number is uint8_t, the number of such threads is reasonably bounded. However, it would not be too difficult to make them exit when idle if preferred.

Comments?

corot, 2018-10-17:
Hi @howardcochran, thank's a lot for your deep analysis of the problem and help on fixing.
I have tried hard to reproduce it with current master code, and failed so far. My guess is that this situation is avoided after commit aaf28cdf1d9d54353b24446a42a36db87c930dc9: dispatching all the callbacks in the single thread started with ros:spin() prevents the concurrency that causes the crash to occur. And so makes TODO unnecessary.
Could you please give a try to current code and see if the crash still happen? Or indeed if running a single dispatch thread has any negative consequence I didn't detect so far.
Thanks again!

howardcochran, 2018-11-02:
@corot, thank you for looking into this. I will, indeed, give that a try and let you know.

spuetz, 2018-11-12:
@howardcochran Did you have the possibility to test that?

howardcochran, 2018-11-14:
@spuetz & @corot, I have tested 0.2.3 (which is the current master), and I believe that the switch to the single-threaded spinner helped.  It is no longer SIGSEGV'ing in the same place.

However, with the recovery_spammer node that I posted above on Oct 4, I now find that it will run the same recovery behavior concurrently even though the client is using the same recovery name and same concurrency slot each time. This causes the recovery behavior to SIGSEGV because it is not expecting to be called concurrently on multiple threads.  I can fix that with a simple mutex in the recovery behavior itself, but I believe that move_base_flex should not call it concurrently when the client used the same concurrency_slot, especially since it canceled the first action before starting a second one.

After adding a mutex to the behavior, now I find that MBF does not handle the cancels correctly. The client starts and cancels recovery actions at a rate of ~10Hz but each invocation of this particular recovery takes several seconds to exit even when .cancel() is called.  Rather than MBF dropping queued up actions that are already canceled, it keeps them queued up and does not attempt to cancel each one in turn until after calling .runBehavior().  This means that it can take hours to drain the queue after killing the spammer client.  During this time, "rostopic echo -n 1 recovery/status" shows many recoveries queued up.

This is one of the things that my proposed change attempts to address.... It will only ever allow one recovery to be "pending" behind the one being preempted in a given concurrency slot.  If third goal is received for this concurrency slot while a second goal is still waiting for the first one to finish canceling, then the second goal is dropped and replaced in the queue by the third one. In my opinion, this behavior needs to be incorporated in some manner.

howardcochran, 2018-11-14:
One more note: I ran the recovery_spammer test against a recovery that takes only a few milliseconds to respond to a cancel. By speeding up the spammer to >100Hz, it still queues up many pending recovery actions and starts them all even though they have already been canceled.

spuetz, 2018-11-14:
@howardcochran  > move_base_flex should not call it concurrently when the client used the same concurrency_slot,
Absolutely! We should set these cases to be rejected. Here we should defensibly wait for the cancel to be finished because the old recovery behaviors has no cancel method. I see your point clearly now! So I'll test your PR again and probably merge it in the next days. But we should also find a way to update running execution objects when a new action goal has been received addressing the execution object. 

And after canceling we should take the latest goal with the same plugin and slot configuration and discard the goals before while using set accepted and rejected.

howardcochran, 2018-11-14:
Yes.  I believe that my PR can be improved quite a bit; let me know what you think.

corot, 2019-07-17:
> Absolutely! We should set these cases to be rejected. Here we should defensibly wait for the cancel to be finished because the old recovery behaviors has no cancel method....

I'm revisiting this issue, and looks like all works as expected after merging #138. But not as @spuetz mentions. Instead, when we run a recovery behavior in a busy concurrency slot, we try to cancel the previous execution and start the new one. For me this is the normal procedure, mimicking a simple action server.

@spuetz, @howardcochran, please reopen the issue if you disagree, or find another problem

closed 2019-07-17 by corot