naturerobots/move_base_flex #96: Calculation needs too much time to stay in the moving frequency!

David9696, 2018-10-17:
Hi,

When I tried to set up the mbf for my robot in gazebo, I got the following warning:

Calculation needs too much time to stay in the moving frequency! (1.772837 > 0.050000)

Then, the robot in simulation would just have intermittent stop and then continue to move. The overall motion is not very smooth. 

May I know what is the problem?

Hope you could enlighten me.

Thanks.

HarshalDawar, 2018-10-17:
The 0.05 that is being displayed represents the (possibly default) value of 20 Hz for the controller frequency. Depending on the your application, you can tone it down to something lower yet sufficient for you. You can set this in multiple ways, E.g: loading a yaml file(say move_base_flex_params.yaml similar to move_base_params.yaml in the move_base setup.

corot, 2018-10-17:
In any case, 1.7728 second for a controller iteration seems like too much.
Do you get the waning occasionally? Or all the time?
In the second case, are you using a custom controller that performs a lot
of calculations?
Otherwise, that could be explained if you slowed down the local costmap
update frequency to less than 1 Hz, as the controller locks the costmap
before calculating robot speed.
Or are you compiling all from sources? If so, can you double check if you
are compiling in Debug?

spuetz, 2018-11-12:
@David9696 any news?

corot, 2019-06-16:
Hi @David9696 , I hope you found the problem. I'll close this issue, as in any case it seems a controller plugin problem, not a MBF-related one.

closed 2019-06-16 by corot