naturerobots/move_base_flex #97: Controller max retries / controller patience question

reinzor, 2018-10-22:
Currently, we can send feedback if a controller fails to generate a valid velocity command. It will retry for x times to find one of for x patience. Using the `controller_max_retries` and `controller_patience` parameters.

This enables waiting for x seconds for a blocked path and then ask for a replan of the planner.

However, I would like to do this only time based and neglect the `max_retries` parameter. This is not possible at the moment (https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/src/abstract_controller_execution.cpp) and the max value of the `retries` parameter is 1000 (https://github.com/magazino/move_base_flex/blob/kinetic/mbf_abstract_nav/src/mbf_abstract_nav/__init__.py#L22).

So if my controller runs on 100hz, my controller patience can max be 10 seconds. What are your thoughts about this?

reinzor, 2018-10-22:
I also noticed that no feedback is being published if the controller fails to generate a valid trajectory. Is this desired behavior?

corot, 2018-10-23:
> For me a value of `-1` would be more logical. Or add an extra enumeration to make this more explicit.

Yes, that's the intended semantics on move_base:

- 0 for no retries at all
- -1 for ignoring and use patience instead

corot, 2018-10-23:
PR [here](https://github.com/magazino/move_base_flex/pull/99). I think this is the expected behavior
max retries:
- > 0 max retries + 1 total attempts
- 0 for no retries at all
- -1 for ignoring and use patience instead
patience:
- 0 ignored (if max retries == -1 will try forever)
- > 0 fail if not command till then

Using isPatienceExceeded() we don't get this behavior, as it checks that there was "any" command, not a valid command, and so it's only exceeded when the controller actually gets more than this time to return. To me this interpretation makes sense, but doesn't seem the one used on move_base

spuetz, 2018-10-23:
I still think that the zero case is confusing. Normally you would expect just one run and then return NO_VALID_CMD if a valid command could not be found.

corot, 2018-10-23:
For max retries == 0? or patience == 0?

spuetz, 2018-10-23:
max retries

reinzor, 2018-10-24:
> I still think that the zero case is confusing. Normally you would expect just one run and then return NO_VALID_CMD if a valid command could not be found.

In case of what @corot suggested, isn't this exactly the behavior you get when you set it to 0? It will try once without a retry so it will give a `NO_VALID_CMD` as a result. In case of a `-1`, it will not give a result based on a retry, and use the patience instead. The outcome will then be provided as feedback.

corot, 2018-10-24:
So can we consider the issue as fixed?

reinzor, 2018-10-24:
Yes, resolved by https://github.com/magazino/move_base_flex/pull/99

closed 2018-10-24 by reinzor