naturerobots/move_base_flex #98: cmd_vel not consistently received since 0.2.1

reinzor, 2018-10-22:
I recently updated to 0.2.2 and the `cmd_vel` output is not consistently received by the subscribers. If I restart a `rostopic echo`, it is received again. Anybody experiencing the same problem?

There have been a lot of thread related commits in the last two releases. I think I isolated the problem to this commit:

102e1cd560c09c291b216b237ab0de9d6a5d6819

When I test the current master with this commit reverted, I do not experience any problems. However, not sure why this commit should tamper with the `cmd_vel` publisher.

corot, 2018-10-22:
Could be related with aaf28cdf1d9d54353b24446a42a36db87c930dc9. I will investigate.

reinzor, 2018-10-22:
I tried this already (this was also my first guess) but problem still seems to be present.

spuetz, 2018-10-23:
I think we should add multiple callback queues with different spinners to exclude this case. I also thought it is because one of the action callbacks needs a lot of time. Before we had a multi thread spinner. @reinzor, So you tried it using the `ros::MultiThreadedSpinner spinner` and the problem still appeared that `cmd_vel` was not published consistently? Did you use only one controller?
Another guess is that in [this line](https://github.com/magazino/move_base_flex/blob/364b230124ea5a0b0f0572d44e40b9eecd84cf52/mbf_abstract_nav/src/abstract_controller_execution.cpp#L78) the publisher is advertised every time a execution object is created which happens on every action call, which is probably not the optimal, right? Advertising and establishing connections can take some time, right?

reinzor, 2018-10-23:
The last point (creating new publishers on a new controller action) sounds like the phenomenon that I am experiencing. Eventually, the connection is always established, however, it might take a couple of seconds.

Has this behavior changed over the last two releases? I don't think it has anything to do with the spinners or callback queues since these only have influence on the subscriptions as far as I know.

corot, 2018-10-23:
Be warned against multi thread spinners! I'm pretty sure issue #93 came from using them, and so could be now solved for free! Hope @howardcochran will have some time to test latest master and confirm my guess!

For the second point, we already publish the goal pose when doing a global plan, and indeed I opened a PR to put the topic under MBF's private namespace:
https://github.com/magazino/move_base_flex/pull/95/files#diff-825aeee832e95b54e73aa9f6fee5a870R54
We can probably drop that topic creation, or do just once at startup

spuetz, 2018-10-23:
Sure #93 is caused by multiple threads running the action callback functions. But for these we could have simply a different callback queue with with one worker thread.

spuetz, 2018-10-23:
And yes, we should maybe make the topic creation once the plugin is loaded.

corot, 2018-10-24:
@reinzor, so reverting to 0.2.0 fixes the problem?

reinzor, 2018-10-24:
Yes, I am currently running this branch: https://github.com/eurogroep/move_base_flex/commits/eurogroep-master Seems to be working as was.

corot, 2018-10-24:
but just reverting to use multi thread spinner doesn't work?

reinzor, 2018-10-24:
I tried this, did not seem to work. I will try to pinpoint this further tomorrow.

reinzor, 2018-10-26:
I tried every commit since 0.2.0 and it seems that this phenomenon occurs since:

https://github.com/magazino/move_base_flex/commit/102e1cd560c09c291b216b237ab0de9d6a5d6819

Any ideas? Anyway, I think it's wise to only construct one command velocity publisher.

spuetz, 2018-10-26:
Looks like a timing side effect.

spuetz, 2018-10-26:
> Anyway, I think it's wise to only construct one command velocity publisher or one per loaded controller.
And yes, I think so, too. I will implement that on Monday.

corot, 2018-10-27:
I'm puzzled...  :no_mouth:, the line added with 102e1cd560c09c291b216b237ab0de9d6a5d6819 takes exactly... 0 seconds. And I cannot appreciate any difference with 0.2.0. So @reinzor, I assume we are using MBF in different ways. Can you provide more details about your usecase:

- How do you call planner and controller?
- Are you calling several controllers in parallel?
- Are you replanning continuously?

And can you measure the time the delete takes? like:

`    ros::Time t = ros::Time::now();
    delete threads_ptrs_[goal_handle.getGoalID().id];
    ROS_ERROR_STREAM("" << (ros::Time::now() - t).toSec());
`
Thanks!

reinzor, 2018-10-27:
I don't think it has anything to do with how long this call takes. I added print statement just before the cmd vel publish and it is publishing just fine. However, it takes time for the subscribers to connect to the publisher. We have a routine where we continuously send exe path goals to the controllers, say once every 2 seconds. Sometimes it takes quite a lot of time for the subscriber to consume the cmd vel of a newly constructed controller (since this happens everytime when we send a new exe path goal).

reinzor, 2018-10-27:
I also can't pinpoint why that specific commit changes the behavior but I think it has something to do with the ros comm internals.

corot, 2018-10-27:
Ok, let's wait for @spuetz to change the code to advertise the topic once, and hope that will solve your issue. By the way, which ROS version are you using?

reinzor, 2018-10-27:
Yep, I'll evaluate next week if ready. We're still on kinetic.

spuetz, 2018-10-30:
@reinzor Could you test again with the current master?

reinzor, 2018-10-31:
Hi @spuetz , the problem seems to be fixed after https://github.com/magazino/move_base_flex/commit/55e860143355188d6a8ec45a1b47b92dbe9e2d2d

Thanks for the update, I will close this issue.

closed 2018-10-31 by reinzor