naturerobots/move_base_flex #100: Global planner planning from map origin instead of robot position

doisyg, 2018-10-25:
Hello,
I am new to move base flex (under melodic) and I am trying to migrate from move_base using mbf_costmap_nav. 
I must be missing something (a param somewhere) but when sending a goal to the robot, the global plan is made from the origin of the map instead of the robot position. I tried with both `global_planner/GlobalPlanner` and `carrot_planner/CarrotPlanner`, same behavior. It seems that the global planner can't get the robot position. Where is it suppose to get the robot frame from ? `/mbf_costmap_nav/global_costmap/robot_base_frame` ?

corot, 2018-10-26:
Interesting...  so you send the goals from RViz? How are you exactly calling MBF?  Thanks

doisyg, 2018-10-26:
Yes goal are sent from Rviz. 
I tried with first with the state machine of the tutorial : http://wiki.ros.org/move_base_flex/Tutorials/Simple%20SMACH%20for%20Move%20Base%20Flex (note: i had to replace `move_base_flex_msgs` by `mbf_msgs` and `move_base_simple` by `mbf_costmap_nav`).
And then with move_base_legacy_relay.py ( again replacing `move_base_flex` by `mbf_costmap_nav`  for the path to the action in the code)
Same result in both case. I believe the controller is fine as after struggling to get to the global path
(although wrong) it could execute it properly.

Below is a screenshot from Rviz where we can see the global path generated by the carrot planner starting at the origin at the map (instead at the position of the robot) and finishing at the goal sent by rviz

[image]

spuetz, 2018-10-26:
Can you check with which frame the goal is published? Is it published from base_footprint? What is the fixed frame you are using?

doisyg, 2018-10-29:
The goal is published in the map frame. I use base_link as the fixed frame (it is the same as base_footprint with a Z offset in my config). See below rostopic output.

```
~$ rosparam get /mbf_costmap_nav/global_costmap/global_frame
map
```

```
~$ rosparam get /mbf_costmap_nav/global_costmap/robot_base_frame 
base_link
```

```
~$ rostopic echo /move_base_simple/goal 
header: 
  seq: 1
  stamp: 
    secs: 196
    nsecs: 747000000
  frame_id: "map"
pose: 
  position: 
    x: 10.391412735
    y: 13.6700534821
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.400012208677
    w: 0.916509810591
---
```

```
~$ rostopic echo /mbf_costmap_nav/move_base/goal 
header: 
  seq: 2
  stamp: 
    secs: 196
    nsecs: 743000000
  frame_id: ''
goal_id: 
  stamp: 
    secs: 196
    nsecs: 743000000
  id: "/move_base-2-196.743"
goal: 
  target_pose: 
    header: 
      seq: 1
      stamp: 
        secs: 196
        nsecs: 747000000
      frame_id: "map"
    pose: 
      position: 
        x: 10.391412735
        y: 13.6700534821
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.400012208677
        w: 0.916509810591
  controller: ''
  planner: ''
  recovery_behaviors: []
---
```

corot, 2018-10-29:
I cannot replicate on Kinetic;  I wonder if it has something to do with switching to tf2 on melodic...
Will try again asa I install melodic

doisyg, 2018-10-29:
I thought about the issue being related to the the switching to tf2 with melodic too.

corot, 2018-11-11:
Right; fixed on PR #106, to be merged soon. Thanks for the debugging effort!

corot, 2018-11-13:
Merged to master; @doisyg, can you confirm that problem is gone on master, so we can close this issue?
Thanks!

spuetz, 2018-11-14:
I'll release today. Thank you.