naturerobots/move_base_flex #102: Feature request: Add cmd_vel to ExePathFeedback

reinzor, 2018-10-29:
For monitoring the robot's intention when a the controller does not return success (due to a collision check for example).

corot, 2018-10-29:
Makes sense; PR always welcome! :neckbeard:

reinzor, 2018-11-01:
I am working on this but it seems that the `current_twist` is implemented incorrectly.

The `current_twist` is filled by `execution.getLastValidCmdVel()`. I though this would be the actual robot velocity. Any thoughts?

howardcochran, 2018-11-02:
Hello.  I'd just like to give my two cents here.  I think that the feedback should show the current_twist that is calculated by the controller (i.e. the commanded velocity), not the actual velocity. For anyone who wants to know the actual velocity, you can subscribe to /odom and get it from there. But the ExePathFeedback is giving feedback from the controller so it seems right that it should say what the controller is commanding.

reinzor, 2018-11-03:
I agree with @howardcochran , maybe it was already intended as this. @corot , could you comment?

corot, 2019-02-25:
Yes, @howardcochran is right, current_twist provides the last calculated command.
If I understood your intention, @reinzor, you want that this twists gets provided as feedback even in case of failure, right? This means moving [this line](https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/src/abstract_controller_execution.cpp#L338) above the _if (outcome_ < 10)_, but given than in the else we call _publishZeroVelocity_, to be coherent the current_twist feedback should be a zero velocity.
Otherwise you would be "lying".

reinzor, 2019-02-26:
That is correct. At least it is nice to see the controller's intention when something is blocking the path for example. However, I think the initial issue came from the fact that the naming is a little bit misleading. `current_twist` sound more like the actual `twist` instead of the `twist_reference` or `twist_command` or `cmd_vel`.

corot, 2019-02-26:
I agree...  specially because it comes together with current_pose, what is actually the current pose of the robot...
I suggest instead last_cmd_vel. @spuetz, stimmt?

reinzor, 2019-05-29:
Covered by https://github.com/magazino/move_base_flex/pull/120

closed 2019-05-29 by reinzor