naturerobots/move_base_flex #105: Problem with costmap when using mbf_costmap_nav pkg

FanhuiJ, 2018-11-09:
Hello everyone,

I want to do some tests with move_base pkg and move_base_flex pkg, so I tried first with move_base, no problem at all. And then, with all the same configurations (like configuration for global and local costmap ect), I tried mbf_costmap_nav package, somehow I got several problems, the robot is not at the center of the local costmap anymore.

I tried to modifed some configurations but it doesn't work, finally I added a local_costmap_params.yaml and loaded it in the move_base_flex.launch file, at this time the robot is at the center of the local costmap, but other problems are still there. When I send a move_base_flex/move_base/goal, I got all these:

My robot couldn't move, even though I re-send a goal manually, it doesn't work:

In fact, I am working on ROSPlan, I tired move_base with rosplan_interface_movebase, it works well. I created an rosplan_interface_movebaseflex, it's similar to rosplan_interface_movebase, just changing the action server name and the goal message type. I think there is no problem with my rosplan interface, just the problem using the mbf_costmap_nav package.

Can anyone help?

Thanks at advance.

corot, 2018-11-11:
Hi, if you are using melodic, I guess you are facing the same problem reported in #100 and fixed with PR #106.
Do you mind to give a try to the PR and report whether this fixes your problem? Thanks!

corot, 2018-11-12:
Fix is now in master, so you can test even easier. Please report here the outcome

FanhuiJ, 2018-11-14:
Thank you for your responses, I re-loaded the updated move_base_flex pkg, there are no more errors when executing the action, only that the move_base_flex/move_base/goal is not sent. When I sent the goal manually, I still got these:

[ERROR] [1542188134.073556855, 274.930000000]: ERROR: failed to set start state

[ INFO] [1542188134.073578100, 274.930000000]: [sbpl_lattice_planner] getting start point (0.00463323,-0.00229892) goal point (3,0)
 

I tried move_base_flex with FlexBE, it's better than in ROSPlan, I don't have the problem with costmap, but I have other problems. I created a move_base_flex_state.py, I found that when I set GlobalPlanner as the planner for the goal, it won't work at all:

[ERROR] [1542188519.209755065, 1423.760000000]: Failed to get a plan.

If I use the SBPLLatticePlanner as the planner for the goal, it works. So I tried this: first using GlobalPlanner, when action failed, change planner to SBPLLatticePlanner, then re-send the goal. At this way, it doesn't work, I got this:

[ INFO] [1542188449.108804409, 1353.800000000]: [sbpl_lattice_planner] getting start point (0.0016118,0.000335229) goal point (2,0)

[ INFO] [1542188449.127952233, 1353.820000000]: Solution not found

I found that I can only use SBPLLatticePlanner, and I need to use it at the very beginning.

spuetz, 2018-11-14:
The topic `move_base_flex/move_base/goal` is the actionlib goal and not the goal pose. Have a look at the [ceres_robot](https://github.com/uos/ceres_robot) package as example. In [ceres_navigation](https://github.com/uos/ceres_robot/tree/master/ceres_navigation) you will find example code for MBF navigation. In [ceres_gazebo](https://github.com/uos/ceres_robot/tree/master/ceres_gazebo) you will find an simulation for the ceres robot. Could you try:

1. `roslaunch ceres_gazebo ceres.launch`
2. `roslaunch ceres_navigation move_base_flex.launch rviz:=true`
3. `rosrun smach_viewer smach_viewer.py`

FanhuiJ, 2018-11-14:
You're right, I can't send a goal manually like this, I have had a look at this package before. I am working on FlexBE now, it's based on smach but writting in other way. I have less problems using mbf_costmap_nav pkg of move_base_flex with flexbe, I have problems with planners.

FanhuiJ, 2018-11-14:
Problems with planners while using move_base_flex with FlexBE solved, problems comes from the wrong configurations.

spuetz, 2018-11-14:
@FanhuiJ All clear? Can we close the issue? Or what do should learn from this? Do we need a configuration tutorial?

FanhuiJ, 2018-11-15:
@spuetz  It ok with FlexBE, I didn't pay attention to maps' resolution. The resolution of the global costmap was different to the resolution of the map (which is published on the topic /map), also I took the wrong motion primitive file of the SBPLLatticePlanner which has a resolution also different to the resolution of the map. After changed the file with the correct resolution and set the same resolution for the global costmap, it works.

But, with these configurations, I still have problems with ROSPlan, when I launch the launch file, I got message : 

terminate called after throwing an instance of 'SBPL_Exception'
  what():  ERROR: failed to read in motion primitive file 

It's the problem with the motion primitive file, not with the move_base_flex package, and now I work on FlexBE, so, we can close the issue. 

Thank you and thank @corot

closed 2018-11-15 by FanhuiJ

spuetz, 2018-11-15:
Okay. One last thing, could you paste the config here for the primitive file? Thank you.

FanhuiJ, 2018-11-15:
@spuetz here is the file:

resolution_m: 0.100000
numberofangles: 16
totalnumberofprimitives: 256
primID: 0
startangle_c: 0
endpose_c: 1 0 0
additionalactioncostmult: 1
intermediateposes: 10
0.0000 0.0000 0.0000
0.0111 0.0000 0.0000
0.0222 0.0000 0.0000
0.0333 0.0000 0.0000
0.0444 0.0000 0.0000
0.0556 0.0000 0.0000
0.0667 0.0000 0.0000
0.0778 0.0000 0.0000
0.0889 0.0000 0.0000
0.1000 0.0000 0.0000
primID: 1
startangle_c: 0
endpose_c: 4 0 0
additionalactioncostmult: 1
intermediateposes: 10
0.0000 0.0000 0.0000
0.0444 0.0000 0.0000
0.0889 0.0000 0.0000
0.1333 0.0000 0.0000
0.1778 0.0000 0.0000
0.2222 0.0000 0.0000
0.2667 0.0000 0.0000
0.3111 0.0000 0.0000
0.3556 0.0000 0.0000
0.4000 0.0000 0.0000
primID: 2
startangle_c: 0
endpose_c: 8 0 0
additionalactioncostmult: 1
intermediateposes: 10
0.0000 0.0000 0.0000
0.0889 0.0000 0.0000
0.1778 0.0000 0.0000
0.2667 0.0000 0.0000
0.3556 0.0000 0.0000
0.4444 0.0000 0.0000
0.5333 0.0000 0.0000
0.6222 0.0000 0.0000
0.7111 0.0000 0.0000
0.8000 0.0000 0.0000
primID: 3
startangle_c: 0
endpose_c: 6 2 0
additionalactioncostmult: 50
intermediateposes: 10
0.0000 0.0000 0.0000
0.0667 0.0222 0.0000
0.1333 0.0444 0.0000
0.2000 0.0667 0.0000
0.2667 0.0889 0.0000
0.3333 0.1111 0.0000
0.4000 0.1333 0.0000
0.4667 0.1556 0.0000
0.5333 0.1778 0.0000
0.6000 0.2000 0.0000
primID: 4
startangle_c: 0
endpose_c: 6 -2 0
additionalactioncostmult: 50
intermediateposes: 10
[... 3750 lines cut ...]
0.0000 0.0000 5.8905
0.0000 0.0000 5.7596
0.0000 0.0000 5.6287
0.0000 0.0000 5.4978
0.0000 0.0000 5.3669
0.0000 0.0000 5.2360
0.0000 0.0000 5.1051
0.0000 0.0000 4.9742
0.0000 0.0000 4.8433
0.0000 0.0000 4.7124
primID: 15
startangle_c: 15
endpose_c: 0 0 2
additionalactioncostmult: 50
intermediateposes: 10
0.0000 0.0000 5.8905
0.0000 0.0000 5.3233
0.0000 0.0000 4.7560
0.0000 0.0000 4.1888
0.0000 0.0000 3.6216
0.0000 0.0000 3.0543
0.0000 0.0000 2.4871
0.0000 0.0000 1.9199
0.0000 0.0000 1.3526
0.0000 0.0000 0.7854

wasih123, 2019-04-10:
Hello, @FanhuiJ , were you able to resolve the "Failed to read in motion primitive file error"? I still get this error even though my resolutions match. Could you please help