naturerobots/move_base_flex #109: Publish zero velocity when reached to goal.

spuetz, 2018-11-19:
Publish zero velocity when reached to goal. If the plugin is not publishing zero velocity to `cmd_vel` the the robot will go one with the last command. This bug is specially caused when setting the `mbf_tolerance_check` to true since the plugin will not publish zero velocity in most of the cases.  Related to #32 and #70.

howardcochran, 2018-11-29:
I request that this be made optional. Some controller provide the ability to arrive at a goal at full speed as a feature.  For example, TebLocalPlanner, when its free_goal_vel parameter set to true. This makes it possible to immediately give it a new goal so that it can go from goal to goal without stopping -- rolling through waypoints.  If MBF hard-coded a zero twist, it would break this capability.

spuetz, 2018-11-29:
Japp you're right!

corot, 2020-02-17:
From #171:

Would it be okay if this becomes the default if controller_->cancel() fails?
If it succeeds it would be up to the user to publishZeroVelocity() if they desire so.

Or perhaps always leave it up to the user to implement this.

I would say the last option is the simplest and more predictable, defaulting to pub the zero vel. @spuetz , your thoughts?
For para name: force_stop_on_cancel = true sounds good?

corot, 2020-05-07:
This got implemented with #166, right?

closed 2020-05-07 by corot