naturerobots/move_base_flex #110: How do we integrate a Python global planner?

hawesie, 2018-11-21:
We have a Python implementation of a global planner that we'd like to use in move_base_flex instead of the existing A* implementation. From the documentation and looking at the running code it appears that we should be able to wrap this in a `GetPath` action to integrate with the rest of the framework. However we can't quite see how to actually configure the system to use a different action server for providing paths. Thanks to @spuetz we are testing this with in the ceres robot (https://github.com/uos/ceres_robot) simulation and nav setup.

corot, 2018-11-21:
> we should be able to wrap this in a GetPath action
exactly. Once done, should not be a problem use it, as long as you use your external executive, like SMACH or BT: the action client on the executive will just send goals to your Python planner node instead of to MBF's node.
Then, you can send the returned plan to any controller, either on MBF node or also external (though in that case is not clear why you need MBF at all!)

hawesie, 2018-11-21:
@corot thanks! I think I see how to integrate this into the existing code now.

hawesie, 2018-11-21:
For future readers, we will be replacing this call https://github.com/uos/ceres_robot/blob/master/ceres_navigation/scripts/planner_sm.py#L31 to our own planner (perhaps just through remapping the name)

closed 2018-11-21 by hawesie

spuetz, 2018-11-28:
So the pluginlib only allows c++ plugins, like shared object libraries. But you can wrap you code with https://www.boost.org/doc/libs/1_46_1/libs/python/doc/ 
You still should implement the interface in mbf_*_core.