naturerobots/move_base_flex #119: mbf doesnt subscribe to my Laserscanner

Salem-Guezguez, 2019-01-28:
I find Mbf very awesome und I just watched your ROSCON video and I did really like it. 
I am now trying to use it on my robot( on Gazebo). I send a goal via Rviz und my robot start to move until the goal is reached. But sometimes my robot makes some collision und doesnt notice that. The robot continue to move toward the wall. I looked at the graph of (rqt_graph) und i found that mbv doesnt subscribe to my topic /scan but the old move_base does. I am really confused!!! Should mfb subscribe to my /scan topic (for my Planner) ?

[image]
[image]
[image]

spuetz, 2019-01-28:
Hey, you should use mbf_costmap_nav instead of mbf_simple_nav, since you need the costmap in your case.

corot, 2019-02-25:
Hi @Salem-Guezguez, and sorry for the late response;
So MBF will listen to whatever sensor is configured in the costmap parameters, for example [here|https://github.com/turtlebot/turtlebot_apps/blob/indigo/turtlebot_navigation/param/costmap_common_params.yaml#L26] is the configured laser for turtlebot. Could be that you are not loading the costmap_common_params.yaml file in your launcher? Please post here the result of:

`rosparam get /move_base_flex`

Salem-Guezguez, 2019-03-12:
Thanks It s work  :))
I have wrong configured  my observation_sources.

corot, 2019-03-12:
Then everybody happy! :ok_man:

closed 2019-03-12 by corot