naturerobots/move_base_flex #121: Transitioning goal to LOST

Salem-Guezguez, 2019-03-27:
Hey , I was recently playing around with the paramters of my planners (dwa and eband) looking for a best combination that leads to a good behaviour and I often faced  these two error messages.
------------------------------------------------------------------------------------------------------------------------
[ERROR] [1553594179.855549955, 32.856000000]: NO PATH!
[ERROR] [1553594179.856440514, 32.856000000]: Failed to get a plan from potential when a legal potential was found. This shouldn't happen.
[move_base_flex-4] process has died [pid 15548, exit code -11, cmd /home/salem/luhbots_ws/devel/lib/mbf_costmap_nav/mbf_costmap_nav __name:=move_base_flex __log:=/home/salem/.ros/log/529d7384-4fab-11e9-824f-dc85def86db3/move_base_flex-4.log].
-------------------------------------------------------------------------------------------------------------------------------
[ INFO] [1553594323.573450136, 162.514000000]: Got new plan
[ INFO] [1553594329.042543536, 167.367000000]: Goal reached
[ WARN] [1553594329.355166851, 167.647000000]: Transitioning goal to LOST
[move_base_flex-4] process has died [pid 18347, exit code -11, cmd /home/salem/luhbots_ws/devel/lib/mbf_costmap_nav/mbf_costmap_nav __name:=move_base_flex __log:=/home/salem/.ros/log/529d7384-4fab-11e9-824f-dc85def86db3/move_base_flex-4.log].
log file: /home/salem/.ros/log/529d7384-4fab-11e9-824f-dc85def86db3/move_base_flex-4*.log
-----------------------------------------------------------------------------------------------------------------------------
the second error always pops up. what is the source of these error? are they instigated by my changing of one of the parameters, one that i shouldn't have?

praveenramanujam, 2019-03-31:
@Salem-Guezguez i had a similar issue but when recovery_behavior was added to the ros parameter, we dont face this issue anymore. the cancel goal seems to look for recovery

corot, 2019-04-07:
The "Failed to get a plan from potential when a legal potential was found. This shouldn't happen." error is caused by [global_planner](http://wiki.ros.org/global_planner). I saw it before, but I have no idea where it comes from. That said, MBF should not crash; I'll investigate it

corot, 2019-06-16:
Solved by #138

closed 2019-06-16 by corot