naturerobots/move_base_flex #122: Seg fault running mbf_costmap_nav

eneiva, 2019-04-03:
I have a strange behaviour running the **`mbf_costmap_nav`**. Whenever I run the node on a virtual machine I have no issues and it is working as intended, however when I run it from my embedded system, it returns instantly a segmentation fault. I have used both the pre-compiled code from the repository and also the source code, and the error persists for both cases. All the other nodes/packages run seamlessly on both systems(embedded system and VM), there aren't presumably networking errors as the other packages are working. `roswtf`gives me nothing.

Both systems are running similar version of ubuntu 16.04, the virtual machine has a x86_64 distro meanwhile the embedded system runs on AARCH64. Both systems are also running the 0.23 version of move_base_flex over ROS kinetic.

The exact line that it seems to trigger the error is the following(in `move_base_server_node.cpp`):
  **`costmap_nav_srv_ptr = boost::make_shared<mbf_costmap_nav::CostmapNavigationServer>(tf_listener_ptr)`**

Both systems have boost 1.58.0. I have thought about what could be so different from both systems but unfortunately I haven't reached any conclusion.

**Output from the terminal of the embedded system:**

The code ran as an example is: **`rosrun mbf_costmap_nav mbf_costmap_nav`**

```
rosrun mbf_costmap_nav mbf_costmap_nav
[ INFO] [1554306259.333460611]: Using plugin "obstacle_layer"
[ INFO] [1554306259.347934963]:     Subscribed to Topics: 
[ INFO] [1554306259.402545142]: Using plugin "inflation_layer"
Segmentation fault (core dumped)
 
```

Running the mbf_simple_nav doesn't trigger this seg fault which shows me at least something from the mbf_costmap_nav ist running. Could it be something related to` tf_listener_ptr`? What could I be missing?

corot, 2019-04-07:
The connection with costmap_2d makes sense, because mbf_costmap_nav uses them, while mbf_simple_nav doesn't.
Can you run mbf_costmap_nav on gdb (how to explained [here](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB)) and paste the stacktrace? Even better, can you recompile in debug and do the same?
Thanks!

eneiva, 2019-04-08:
Recompiled the workspace with -DCMAKE_BUILD_TYPE=Debug

```
nvidia@nvidia:~/catkin_ws$ rosrun --prefix 'gdb -ex run --args' mbf_costmap_nav mbf_costmap_nav 
GNU gdb (Ubuntu 7.11-0ubuntu1) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/nvidia/catkin_ws/devel/lib/mbf_costmap_nav/mbf_costmap_nav...done.
Starting program: /home/nvidia/catkin_ws/devel/lib/mbf_costmap_nav/mbf_costmap_nav 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fb44a1120 (LWP 5690)]
[New Thread 0x7fb3ca1120 (LWP 5691)]
[New Thread 0x7fb34a1120 (LWP 5692)]
[New Thread 0x7fb2ca1120 (LWP 5693)]
[New Thread 0x7fb24a1120 (LWP 5694)]
[New Thread 0x7fb1ca1120 (LWP 5695)]
[New Thread 0x7fb14a1120 (LWP 5696)]
[New Thread 0x7fb0ca1120 (LWP 5697)]
[ INFO] [1554713705.534366709]: Using plugin "obstacle_layer"
[ INFO] [1554713705.602884899]:     Subscribed to Topics: 
[ INFO] [1554713705.662088021]: Using plugin "inflation_layer"
[New Thread 0x7f93fff120 (LWP 5698)]
[New Thread 0x7f937ff120 (LWP 5699)]

Thread 11 "mbf_costmap_nav" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f937ff120 (LWP 5699)]
0x0000007fb7b99674 in ros::Time::now() () from /opt/ros/kinetic/lib/librostime.so
(gdb) bt full
#0  0x0000007fb7b99674 in ros::Time::now() () from /opt/ros/kinetic/lib/librostime.so
No symbol table info available.
#1  0x0000007fb718fc98 in costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#2  0x0000007fb7190db8 in costmap_2d::Costmap2DROS::updateMap() () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#3  0x0000007fb7191a34 in costmap_2d::Costmap2DROS::mapUpdateLoop(double) () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#4  0x0000007fb6e0ecf8 in ?? () from /usr/lib/aarch64-linux-gnu/libboost_thread.so.1.58.0
No symbol table info available.
#5  0x0000007fb7039fb4 in start_thread (arg=0x7fb6e0ec30) at pthread_create.c:335
        pd = 0x7fb6e0ec30
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {547935482144, 549755803736, 0, 549755803727, 548529040432, 548531462144, 548485736176, 7754544, 1, 547935482144, 547935480080, 13568342539068514507, 
                0, 13568342538606892639, 4617315517961601024, 4591870180066957722, 4576918229304087675, 0, 0, 0, 0, 0}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, 
              cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#6  0x0000007fb78f9390 in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:89
No locals.
(gdb) s
Single stepping until exit from function _ZN3ros4Time3nowEv,
which has no line number information.
[ERROR] [1554713715.733687113]: poll failed with error Interrupted system call
[Thread 0x7fb34a1120 (LWP 5692) exited]
[Thread 0x7fb3ca1120 (LWP 5691) exited]
[Thread 0x7f937ff120 (LWP 5699) exited]
[Thread 0x7f93fff120 (LWP 5698) exited]
[Thread 0x7fb0ca1120 (LWP 5697) exited]
[Thread 0x7fb14a1120 (LWP 5696) exited]
[Thread 0x7fb1ca1120 (LWP 5695) exited]
[Thread 0x7fb24a1120 (LWP 5694) exited]
[Thread 0x7fb2ca1120 (LWP 5693) exited]
[Thread 0x7fb44a1120 (LWP 5690) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.

```

corot, 2019-04-08:
Yup, on costmap_2d. Did you try to use old good move_base on your ARM ? I would expect you have the same problem. If so, probably the place for this issue is [ROS navigation repo](https://github.com/ros-planning/navigation)

eneiva, 2019-04-08:
You are right @corot, move_base also triggers seg fault I have opened a [issue](https://github.com/ros-planning/navigation/issues/883) on the ROS nav repo.

closed 2019-04-08 by eneiva