naturerobots/move_base_flex #126: Getting descriptive outcomes from ExePathAction

amrp1, 2019-05-15:
Thank you for this great package. 

I am trying to design some recovery behaviors for a bot following a set of waypoints, with velocities issued by TEB. I am using ExePathAction to start the navigation with TEB. 

I would like to know when the path is blocked (BLOCKED_PATH = 109) in order to trigger either a GetPathAction until the next waypoint, or maybe some other recovery behaviour. 

Are the descriptive outcomes + messages already implemented in MBF? Or do these descriptive outcomes have to be implemented in the controller (in this case: TEBLocalPlannerROS)?
Or should it be implemented by extending some function of the controller_action or abstract_controller_execution?

Thank you for your time, any help will be appreciated! :)

corot, 2019-05-15:
Hello @amrp1, this sounds much more like a [ROS Answer](https://answers.ros.org/questions/), as github issues are mainly to report bugs or request new features. Moreover, much more people will benefit for the answers and comments posted.

I promise I'll answer there soon after you post the question :crossed_fingers: 
Please paste here the URL of the question

amrp1, 2019-05-15:
HI @corot, here's the [question on ROS Answers](https://answers.ros.org/question/323207/getting-descriptive-outcomes-from-exepathaction/). I'll close the issue here.

closed 2019-05-15 by amrp1

corot, 2019-05-15:
Answered [here](https://answers.ros.org/question/323207/getting-descriptive-outcomes-from-exepathaction/)