naturerobots/move_base_flex #129: No costmaps published with new plugin

amrp1, 2019-05-27:
I replaced the Global Planner with a version that derives from mbf_costmap_core. When I try to launch mbf_costmap_nav with the modified global planner loaded as a plugin, there are no global/local costmaps being published. Is this a bug? Am I missing something?

I'm using Kinetic on Ubuntu 16.04.

corot, 2019-05-27:
For sure this is not an intended behavior... Can you check the value of local_planner/update_frequency and local_planner/publish_frequency parameters?

amrp1, 2019-05-27:
`rosparam get /mbf_costmap_nav/TebLocalPlannerROS/controller_frequency` returns 20.0, but the local_planner/update_frequency and local_planner/publish_frequency parameters are not set.

corot, 2019-05-27:
This sounds to me like a configuration problem. Can you post the question in ROS answers, and provide more details? Like 
- What are you running exactly?
- What's your configuration?
- What can you see in RViz?

closed 2019-05-27 by corot

amrp1, 2019-05-27:
I was having some issues with the plugin and so I reinstalled the global_planner and move_base_flex and the costmaps are being published now.

> local_planner/update_frequency and local_planner/publish_frequency parameters
by this, did you mean local_costmap/update_frequency and local_costmap/publish_frequency? These parameters return 25.0 and 1.0 respectively. 

I believe that the errors had something to do with version mismatches between kinetic and melodic as mentioned in #774. I made the changes in the melodic version of the global planner instead of the kinetic one.