naturerobots/move_base_flex #130: Release plans

reinzor, 2019-05-29:
Could you provide some insight in when you are planning to do a 1.0.0 release and from there ensure ABI compatibility over ROS versions?

reinzor, 2019-06-04:
ping

spuetz, 2019-06-04:
Hi I was just about to answer :D So we will ensure API compatibility over two minor version changes, and we'll make things deprecated if necessary and remove these after two minor version changes. Thereby one will have enough time to adapt to the new / not deprecated API. Minor versions are counted when new features come in. This is how we version MBF.

spuetz, 2019-06-04:
@reinzor Btw. Since we have new navigation servers based on MBF, like one for outdoor terrain working on meshes, and on in development working on grid_map. Cheers.

reinzor, 2019-09-02:
Thanks for the update; I was more referring to https://semver.org/#how-do-i-know-when-to-release-100 , but thanks for ensuring api compatibility :)

closed 2019-09-02 by reinzor