naturerobots/move_base_flex #131: Sending multiple GetPathAction Goals using Actionlib

amrp1, 2019-06-04:
I'm using `move_base_flex` to plan between waypoints using GlobalPlanner. The reason I use this instead of linearly interpolating between points is to be able to avoid obstacles while planning. I've written a function that sends a GetPathGoal and tracks it until a result is received, and then proceeds to do so for the rest of the waypoints. The idea is to concatenate all the received plans and send them using ExePathAction. At first I thought the costmaps were not being taken into consideration. Until the first point, it is able to use the costmaps to plan around obstacles, but can't do that with the rest of the waypoints (interpolates linearly). I can't figure out why this happens. Does it have something to do with the concurrency slots or the logic of the code? 

```c++

std::vector<geometry_msgs::PoseStamped> MBFNav::interpolateGlobalPlanPath(nav_msgs::Path& path)
{
  
  std::vector<geometry_msgs::PoseStamped> global_path_;

  //call next waypoint service
  get_point_client.call(get_next_point_msg);
  next_waypoint_index = get_next_point_msg.response.index;
  //ROS_INFO("next point index %i",next_waypoint_index);

  //populate get_path_goal_ for current position to next point
  get_path_goal_.use_start_pose = false;
  get_path_goal_.target_pose = path.poses[next_waypoint_index];
  get_path_goal_.tolerance = 0.1;
  get_path_goal_.planner = "GlobalPlanner";
  get_path_goal_.concurrency_slot = 0;
  //ROS_INFO("next point x: %f y: %f",path.poses[next_waypoint_index].pose.position.x,path.poses[next_waypoint_index].pose.position.y);
  
  //call GetPath action with get_path_goal_
  gpc.sendGoal(get_path_goal_);
  ROS_INFO("fetch plan to first point");

  gpc.waitForResult();
  ROS_INFO("planning from current pose to first waypoint");  

  //get global plan result
  mbf_msgs::GetPathResultConstPtr start_result = gpc.getResult();
  ROS_INFO("planning outcome %s",start_result->message.c_str());
  
  //add to global_path_
  global_path_ = start_result->path.poses; 
   
  //for loop : GetPath between rest of the points on path and populate global_path_
  for (int i = next_waypoint_index ; i< path.poses.size() -1 ; ++i){
        //populate get_path_goal_ for current position to next point
        temp_get_path_goal_.use_start_pose = true;
        temp_get_path_goal_.start_pose = path.poses[i];
        temp_get_path_goal_.target_pose = path.poses[i+1];
        temp_get_path_goal_.tolerance = 0.1;
        temp_get_path_goal_.planner = "GlobalPlanner";
        get_path_goal_.concurrency_slot = 0; //commenting this line out made no difference
        //call GetPath action with get_path_goal_
        gpc.sendGoal(temp_get_path_goal_);

        gpc.waitForResult();
        ROS_INFO("Waiting for path in loop");

        // get global plan result
        mbf_msgs::GetPathResultConstPtr temp_result = gpc.getResult();
        ROS_INFO("point number:%i message: %i",i,temp_result->outcome);
        if (temp_result->outcome == mbf_msgs::GetPathResult::PAT_EXCEEDED){
          continue;//plan to next point
        }
        else{
        global_path_.push_back(temp_result->path.poses.back());
        }
  
  }  

  return global_path_;

}

```

amrp1, 2019-06-04:
deleted

closed 2019-06-04 by amrp1

corot, 2019-06-04:
:thinking: