naturerobots/move_base_flex #143: Using MBF with FlexBE to send ExePathGoal

amrp1, 2019-07-08:
I am using the move_base_flex package with FlexBE, to navigate a mobile robot through a set of waypoints. To do this I have to construct a composite global plan from other planners and then pass the path to ExePath action. When I try to pass the new ExePath action to the goal, it throws the exception `'list' object has no attribute 'header'`. I'm using the nav_msgs/Path message to pass on the global plan. I made sure to fill out the frame_id, and stamp of the Path message. All individual PoseStamped messages also have the header included. What am I missing? 

I am using Ubuntu 16.04 with ROS Kinetic, FlexBE and Python.

```
def on_enter(self, userdata):

  goal = ExePathGoal()
  goal.path.header.stamp = rospy.Time.now()
  goal.path.header.frame_id = "map"
  goal.controller = self.controller
  goal.concurrency_slot = self.concurrency_slot

  userdata.path_out = userdata.path

  self.error = False
  try:
      self.client.send_goal(self.topic, goal)
  except Exception as e:
      Logger.logwarn('Failed to send the ExePath command: \n%s' % str(e))
      self.error = True

def on_exit(self, userdata):
      if not self.client.has_result(self.topic):
      self.client.cancel(self.topic) 
      Logger.loginfo('Cancelled active ExePath action goal')

```

corot, 2019-07-09:
mmm...  as it is now, the error you report is "impossible", he he.  Can you send me the full script, so I try to reproduce? I suppose this is the breaking line, right?
`goal.path.header.stamp = rospy.Time.now()`

amrp1, 2019-07-15:
There was a problem with how I was combining the paths. I tried to `extend` the existing PoseStamped array in a message, with another PoseStamped array, instead I had to `append`. So this has been resolved. Apologies for the delayed update.

closed 2019-07-15 by amrp1