naturerobots/move_base_flex #144: mbf_costmap_nav crashed when homotopy is enabled

27Apoorva, 2019-07-08:
Hi,
I was trying to use MBF with TEB Local planner. TEB offers a parameter to enable homotopy. If in file `teb_local_planner_params.yaml` , I set `enable_homotopy_class_planning: True`, I get the following from move base flex.
```
[ WARN] [1562594426.753771999]: Trying to stop the planning rigorously by interrupting the thread!
[ WARN] [1562594426.821746017]: Controller thread interrupted!
[ WARN] [1562594426.821940230]: The controller has been stopped!
[ WARN] [1562594432.059184930]: Transitioning goal to LOST
[ERROR] [1562594432.059362015]: mbf_costmap_nav crashed (signal: 11)
```

@corot Any help would be appreciated. Thanks.

corot, 2019-07-09:
Are you using master branch? Or debs? If debs, kinetic or melodic. And can you check if you are using latest version 0.2.4?

The first message comes from the fact that TEB gets much, much slower when enabling homotopy class planning, and so takes more than the default controller patience (1 second, I think).

The transition goal to LOST and crash could be fixed on release 0.2.4, but not 100% sure, so I need to know what code are u using

27Apoorva, 2019-07-09:
@corot Hi. Thanks for your reply. I am using 0.2.4 using your master branch. I knew that there was an issue before of mbf crashing on lost goals. Also, if you could suggest how to improve homotopy parameters so that I don't get this warning. My robot doesn't even move if I turn on homotopy.

corot, 2019-07-10:
I get a different outcome when TEB takes longer than patience:
```
[/navigation/move_base_flex  INFO 1562757670.255449424]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757670.255576683]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757670.755823603]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757670.755949883]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757671.256233292]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757671.256403320]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757671.756701644]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757671.757665650]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757672.258434894]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757672.258578318]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757672.758799328]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757672.758922643]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757673.259241968]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757673.260205910]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  INFO 1562757673.760487646]: The controller patience has been exceeded! Stopping controller...
[/navigation/move_base_flex  WARN 1562757673.761745160]: Trying to stop the planning rigorously by interrupting the thread!
[/navigation/move_base_flex  WARN 1562757674.538156996]: The controller has been stopped!
```
And the action client gets a None result. No LOST goal, no crash. No idea from where comes the difference. Can you retry running MBF with gdb and print the stacktrace?

As for the homotopy parameters, problem seems to be having too many obstacles. If you are using costmap_converter plugins, tweak params to make clustering into polygonal obstacles much more aggressively, e.g.:
```
CostmapToPolygonsDBSMCCH:
    cluster_max_distance: 0.75
    cluster_min_pts: 2
    cluster_max_pts: 200
    convex_hull_min_pt_separation: 0.1
```

27Apoorva, 2019-07-11:
@corot Not sure how but I resolved the issue of crash. Some weird cached files from some other version was causing it I guess. For the homotopy issue, I am unable to load costmap conversion plugin. I followed the tutorial but somehow it is not getting loaded for me. Even the tutorial is failing. I changed few things in code and I am using it as a node now. Which seems to work for me.

corot, 2019-07-12:
Good you solved the problem. So I can close this issue, right? Feel free to reopen otherwise.

closed 2019-07-12 by corot