naturerobots/move_base_flex #145: Exe_path with continous replanning

MarcelSoler, 2019-07-22:
Hi! I am setting a behaviour tree with move_base_flex, I wanted to be able to do continuous replanning while calling GetPath and ExePath. I am using the blackboard to share the path between them. 

Now I am working on calling GetPath al low freq (1/2Hz) and ExePath  higher ( 20Hz). Getting a Behaviour similar to move_base action but with more control in the logic, recoveries ...

 In #70 something similar was mentioned and I am not sure if this was resolved. 

Can someone clarify this ? or suggest a way to do it! 

Thank you,

corot, 2019-07-22:
Hi! yes, all is working fine. We do what you are describing also with a BT:

[image]

Sending a new goal to exe_path while already running will just replace the executed path, without stopping/starting the robot, as long as you use the same concurrency slot and the same controller.

MarcelSoler, 2019-07-22:
Nice! really nice! thank you for your quick response.
FYI I'm using https://github.com/BehaviorTree/BehaviorTree.CPP and it's very powerful and easy to use.

closed 2019-07-22 by MarcelSoler

artemiialessandrini, 2020-08-31:
Thanks for sharing a link to BT, @MarcelSoler

juliangaal, 2021-03-24:
> Hi! I am setting a behaviour tree with move_base_flex, I wanted to be able to do continuous replanning while calling GetPath and ExePath. I am using the blackboard to share the path between them.

Out of curiosity: Is there any reason to do this manually instead of playing with `planner_frequency`?

Rayman, 2021-03-24:
@artemiialessandrini smach is currenly not maintained very well. There are multiple PR with no reactions. I would look for a different framework.

I've experience with smach, it works, however for new projects I'm also looking at BehaviorTree.CPP. Looks very nice.