naturerobots/move_base_flex #146: Oscillation detection doesn't take into account in-place rotation

nicoluce, 2019-07-23:
Currently oscillation timer is reset when the robot traverses a minimum distance, but if the robot performs an in-place rotation MBF considers that as oscillation. 

https://github.com/magazino/move_base_flex/blob/62b46461fd2a04f10dd33f0f9f947ce33baea193/mbf_abstract_nav/src/move_base_action.cpp#L206

Having an `oscillation_angle` parameter and checking the angle difference with that can solve the issue.

corot, 2019-07-23:
That's true. Admitedly, we never bother too much about the move_base action, just replicate old good move_base behavior plus an extended interface. So we replicate [this (possible wrong) behavior](https://github.com/ros-planning/navigation/blob/melodic-devel/move_base/src/move_base.cpp#L806).

Do you think it makes sense to have add that check? I mean, a robot just spinning is most probable stuck, and so we want to react. I suppose this is the rational for not including angle check in the first place.

nicoluce, 2019-07-23:
You're right about the spinning. For the time being I can augment the timeout to a value greater than the time it takes the robot to make an 180 turn, but personally I think there should be a better way to handle it.

corot, 2019-08-30:
Hi @nicoluce, I'll close this issue, as we will by now stick to the old move_base behavior.
Feel free to propose a better alternative, possible on https://github.com/ros-planning/navigation, to open a discussion on the issue.

closed 2019-08-30 by corot