naturerobots/move_base_flex #147: Oscillation timeout doesn't reset after recovery succeeded

nicoluce, 2019-07-23:
This happens when using **Move Base action**. 
When oscillation is detected and recovery behaviors are executed, MBF doesn't give enough time to the robot to move and get out of the oscillating state. 

This causes that all recovery behaviors are fired and when none are left the goal is aborted. 
https://github.com/magazino/move_base_flex/blob/62b46461fd2a04f10dd33f0f9f947ce33baea193/mbf_abstract_nav/src/move_base_action.cpp#L240

`last_oscillation_reset_` and `last_oscillation_pose_` could be reset here and solve the issue.
https://github.com/magazino/move_base_flex/blob/62b46461fd2a04f10dd33f0f9f947ce33baea193/mbf_abstract_nav/src/move_base_action.cpp#L474

corot, 2019-07-27:
Thanks, you are right. Please try PR #148. Should fix the issue

nicoluce, 2019-07-29:
:+1: Works for me. Thank you.

commit 61b9efee (2019-07-29): Fix #147: Reset oscillation timer after executing a recovery behavior (#148)

* Reset oscillation timer after executing a recovery behavior

* Do not mention recovery behaviors on oscillation failure on move_base action
closed 2019-07-29 by spuetz