naturerobots/move_base_flex #149: Local planner

abylikhsanov, 2019-08-07:
I have specific kinematics for my robot. In order to save time, do you know whether it is possible to change the kinematics and trajectories without touching any core classes?

corot, 2019-08-13:
Hi, mbf has nothing to do about robot kinematics; those are handled at local and global planners. I would rather post the question in ROS answers, but please add more details about your actual problem and intentions.

abylikhsanov, 2019-08-13:
Hi,

Yes I do know the interaction between the original move base and planners,
where the planners construct the trajectory and move base sends velocity
commands to complete the trajectory. I was asking more about the API use of
this package and whether you have any examples of using the api  (planners
to move base flex)

corot, 2019-08-19:
Well, the planners API are defined [here](https://github.com/magazino/move_base_flex/tree/master/mbf_abstract_core/include/mbf_abstract_core). As you see, there aren't big changes compared to nav_core API; mainly we add return code an error message to better react to unexpected situations.

I added support for both nav_core and mbf_costmap_core APIs on [my ROS navigation fork](https://github.com/corot/navigation/tree/use_mbf_core). I also did so for [TEB local planner](https://github.com/rst-tu-dortmund/teb_local_planner/commit/484fd2c8073330b121469b1b3bd28a35f0abbf8a).

corot, 2019-08-30:
Hi @abylikhsanov, I'll close this issue as solved. Please post a question in ROS answers if you still need help on using your local planner with MBF.

closed 2019-08-30 by corot