naturerobots/move_base_flex #152: Problem with clearing costmap recovery

MarcelSoler, 2019-08-28:
Hi!
I am using  recovery_action_client to trigger a clear_costmap_recovery. 
The action_client works properly returning succeed. At the move_base_flex node I receive:
```
 Start action "recovery" using recovery "clear_costmap" of type "clear_costmap_recovery/ClearCostmapRecovery"
[ WARN] [/move_base_flex]: Clearing costmap to unstuck robot (3.000000m).
```
All seems to be working properly however  the global_costmap is **not** cleared.
If I do it manually with `rosservice call /move_base_flex/clear_costmaps "{}"`
It works fine.

My recovery_behaviors file:
```
recovery_behaviors:
  - name: 'clear_costmap'
    type: 'clear_costmap_recovery/ClearCostmapRecovery'

# clear_costmap:
#  reset_distance: 0.0
```

Am I doing something wrong? 
Any help would be appreciated.

nlimpert, 2019-08-29:
Maybe related to https://github.com/ros-planning/navigation/pull/641?

MarcelSoler, 2019-08-30:
I don't think so, I even tried to specify the layers I wanted to clear...

corot, 2019-08-30:
Hi, I cannot reproduce the problem, as u can see in this sequence:

[image]

Please check the configuration of the clear_costmaps recovery behavior:

`rosparam get /move_base_flex/clear_costmaps`

Could be that you have configured it to clear only local costmap? This option is available after https://github.com/ros-planning/navigation/commit/ec0ef1e741480be5b370d91e3e1cb612f2d2bf97

Also note that the clear will take some time, depending on the update frequency of your global costmap, and won't take place if its zero.

MarcelSoler, 2019-08-30:
```
rosparam get /move_base_flex/clear_costmaps
ERROR: Parameter [/move_base_flex/clear_costmaps] is not set
```
Mmmm strange... where is this parameter set?? can you send me an example. Some time ago I used ceres_robot as example

corot, 2019-08-30:
Well, if you don't have any parameter defined for the clear_costmaps, means u are using the default configuration and so it should clean both costmaps. Btw, I assume your MBF node is called move_base_flex, what can be not the case. Could be also mbf_costmap_nav?  Then change to 

`rosparam get /mbf_costmap_nav/clear_costmaps`

Take a look also at global_costmap/update_frequency, to ensure it's not 0

MarcelSoler, 2019-09-02:
Update:

I filled all the options just to be sure:

### recovery_behaviors.yaml:
```
recovery_behaviors:
  - name: 'clear_costmap'
    type: 'clear_costmap_recovery/ClearCostmapRecovery'

clear_costmap:
  reset_distance: 0.0
  force_updating: true
  affected_maps: both
  layer_names: obstacles_laser rgbd_obstacle_layer
```

### Check ros_params:
```
rosparam get /navigation/move_base_flex/global_costmap/update_frequency
3.0
rosparam get /navigation/move_base_flex/clear_costmap/affected_maps
both
rosparam get /navigation/move_base_flex/clear_costmap/layer_names
obstacles_laser rgbd_obstacle_layer
rosparam get /navigation/move_base_flex/clear_costmap/force_updating
true
rosparam get /navigation/move_base_flex/clear_costmap/reset_distance
0.0
rosparam get /navigation/move_base_flex/clear_costmaps
ERROR: Parameter [/navigation/move_base_flex/clear_costmaps] is not set

```
### results

#### move_base_flex terminal:
```
 Start action "recovery" using recovery "clear_costmap" of type "clear_costmap_recovery/ClearCostmapRecovery"
[ WARN] [/navigation/move_base_flex]: Clearing costmap to unstuck robot (0.000000m).
```

#### goal topic:
```
---
header:
  seq: 48
  stamp:
    secs: 1567418684
    nsecs: 179497169
  frame_id: ''
status:
  goal_id:
    stamp:
      secs: 1567418684
      nsecs: 178718466
    id: "1"
  status: 3
  text: "Recovery \"clear_costmap\" done"
result:
  outcome: 0
  message: "Recovery \"clear_costmap\" done"
  used_plugin: ''

```
It doesn't work, both costmaps don't even flicker...

corot, 2019-09-02:
Interesting....  But obstacles_laser rgbd_obstacle_layer are not standard
costmap layers. What if you let this parameter empty? Or change to just
obstacle_layer?

MarcelSoler, 2019-09-02:
I've renamed my obstacles_laser layer to obstacle_layer, and removed the layer_names param,
I also tried with ` layer_names: obstacle_layer `  same result...

The only way to clear costamp that's working in my robot is with the clear_costmap service.

corot, 2019-09-02:
And removing all clear_costmap parameters?

Also you can try to call the recovery action directly, setting
'clear_costmap' on field 'behavior', e.g. using:

rosrun actionlib axclient.py /navigation/move_base_flex/recovery

MarcelSoler, 2019-09-02:
Sorry to bother you so much... 

I've removed all clear_costmap parameters.

[image]

```
Start action "recovery" using recovery "clear_costmap" of type "clear_costmap_recovery/ClearCostmapRecovery"
[ WARN] [/navigation/move_base_flex]: Clearing costmap to unstuck robot (3.000000m).
```
Still it didn't clear the costmap. 

FYI I'm using compiled mbf at melodic branch on latest commit.

corot, 2019-09-04:
I'm completely lost... Could be that the costmap indeed gets cleaned but
it's not reflected on rviz? Try:

rosparam get /navigation/move_base_flex/global_costmap/publish_frequency

Otherwise, I think you have no option but debugging clear costmap recovery
itself

MarcelSoler, 2019-09-10:
I'm also a bit lost...
The costmap doesn't get cleaned, I have publish_frequency of 1Hz and when I manually (with the service) clean the costmap I can see the changes.

I'll try to debug the clear costmap recovery but It worked for me with move_base and I have almost the same parameters, is there some kind of documentation about mbf params.

spuetz, 2019-09-10:
Can you print the debug outout here. Thank you. Where do you call the recovery action? Or do you use the move_base action? Pleas give us some details.

spuetz, 2019-09-10:
You could compile the clear costmap plugin and figure out what is represented by https://github.com/ros-planning/navigation/blob/d60209b348add7db379a50aaac3de8af3c7716e7/clear_costmap_recovery/src/clear_costmap_recovery.cpp#L142
name should be in `clearable_layers_` ... If you find something interesting  let us know ;) Thank you.