naturerobots/move_base_flex #154: Cancel controller plugin with TEB

27Apoorva, 2019-09-11:
Hi,
I am using move_base_flex with TEB planner. 
For continuous replanning, we swap the pointer to the controller and sets cancel on the old goal. This results in "Goal preempted by a new plan" message and goal status as Preempted. When the cancel is called on the controller itself, the message is "Controller canceled" on the exe_path result topic and status is still Preempted. Since TEB doesn't provide cancel control action, I added `execution.stop()` to stop the current thread and I publish a zero velocity command. In my behavior tree whenever a controller fails (either due to cancel or aborted goal) I go into recovery mode. The bug I found was that when the goals are cancelled, the recovery and controller are both running together publishing two different velocities. I don't think I have any issue in my behavior tree. I think the cancel behavior is what needs proper implementation. Any suggestions what should it be?

corot, 2019-09-14:
Hi, all that should be superfluous: if you send a new goal to exe_path with the same plugin and concurrence slot (or just default values), but a different plan, MBF will automatically replan, and the old goal will end on PREEMPTED, if I remember properly.
That's the way we use both TEB and DWA.

Please tell me if all works!

27Apoorva, 2019-09-16:
I get what you said. I understand that. I am switching the controller from TEB to DWA and TEB's goal ends on CANCELLED which calls `AbstractControllerExecution::cancel()` function. In `controller->cancel()`, for both TEB and DWA I implemented a cancel function which publishes zero velocity command. My problem is that when TEB gets CANCELLED and DWA starts running, and then lets say next we want to switch back to TEB, so DWA now gets CANCELLED, for some reason by controller is still running publishing zero velocities.

corot, 2019-09-18:
So you mean you want to switch controllers alternatively, right? I have never tried that... The expected behavior is what you describes in the first witch from TEB to DWA: cancel the currently running, then starting the other. What you describe next is unexpected, I need to debug it. I'm currently on a trip, though, so not sure when I'll find time, sorry.

27Apoorva, 2019-09-18:
Sure. No problem. Yes The behavior on switching controllers is unexpected. If you can try it out and let me know the ideal behavior, that would be great. No worries on timeline. For now I figured out a solution (not a good design) but solves it for now.

27Apoorva, 2019-09-30:
@corot Please let me know if you got some time to be able to try this out. Thanks

corot, 2019-10-01:
> @corot Please let me know if you got some time to be able to try this out. Thanks

not yet, sorry. And wont in at least a week :crying_cat_face:

corot, 2019-10-07:
Hello back; I found some time to test this issue. I'm switching many times during navigation between the 2 controllers packed with ROS navigation stack (`Trajectory Rollout` and `DWA`) without MBF interface and all works as expected: each switch I get the WARN _"Cancel controlling failed or is not supported by the plugin..._, expected because neither implement cancel method. But all is fine, the controllers are switched as soon as the current call to calculate velocity ends (in around ~50 ms when running controllers at 10 Hz as I do).
No reason why TEB should be different, but I'll try anyway.

So questions:

- Did you try switching controllers without any cancel implementation nor changes on MBF? Please try if not; my expectations are that it should work.
- Are you using the controller plugins implementing the MBF interface (mbf_costmap_core instead of nav_core)?

corot, 2019-10-24:
Hello back; I made a similar test switching between TEB and Trajectory Rollout with MBF interfaces and  works same way as in my previous experiment. @27Apoorva, can you try switching controllers without any cancel implementation nor changes on MBF? Please try if not; my expectations are that it should work.

27Apoorva, 2019-10-24:
Hi I will try it but the problem I was telling is not something that happens too frequently. Basically, I was switching the controllers when let's say a person walks in front of the robot. And I didn't see this issue always but only sometimes. I would just want to know that let's say you are using DWA for normal navigation, a person/obstacle walks in front of the robot and you want to go into recovery, in that case DWA and Recovery plugin both starts publishing velocity. I have seen it couple of times but not always. May be I can remove cancel implementation and try and update it it fixes the issue. This was the actual issue. Sorry for making it so confusing.

corot, 2019-10-24:
To avoid both DWA and Recovery plugin publishing velocity in the case you describe, you just need to cancel DWA, that is, exe_path action, and then run the recovery behavior. This works bullet-proof, as we do it continuously with our robots. And you don't need to publish a zero velocity command, as I added this to master this week! So yes, please try to remove your cancel implementation.

27Apoorva, 2019-10-25:
Thank you. I just wanted to be sure this is how it needs to be. My fix to stop this from happening was to cancel the exe_path action and publish zero velocity from TEB/DWA planner. I think both are almost the same so should be fine. I am closing it out. Thanks for all the clarification.

closed 2019-10-25 by 27Apoorva