naturerobots/move_base_flex #157: dist_to_goal & angle_to_goal while continuous replanning

karan-narula, 2019-09-23:
Hey guys,

I have implemented an intermediate planner which generates a sub-goal based on the global plan. This plan is sent to the controller and is continuously updated (much like #145). Your implementation of updating the plan (invoking `setNewPlan`) when `ExePath` action is called on the same concurrency slot works fine (discussed in #70). 

However, I have a problem with the feedback. From what I can see in the implementation (in controller_action.cpp), the  `goal_pose_`, which takes the last position of the plan, is only updated once even when continuously re-planning. This makes the `dist_to_goal` and `angle_to_goal` part of the feedback not correct in my application as the sub-goal is moved toward the real goal when the robot moves. I would very much like the `goal_pose_` to take the last position of the plan every time a new plan is given. I wonder if the current implementation is part of your design philosophy or if you had any plan of including this flexibility. If there is a way to achieve this functionality without modifying upstream code, i.e. a work-around, please do let me know.

Thank you in advance for any help and great work on the package! I am happy to elaborate if the explanation of the issue is in any way unclear.

corot, 2019-09-29:
Hi, thanks for the feedback. Actually, I would say that current behavior is a bug, and that goal_pose_ should get updated on replanning. Trivial fix (I didn't try) would be to add:

`goal_pose_ = goal_handle.getGoal()->path.poses.back();`

in  https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/src/controller_action.cpp#L80

Please feel free to try and report if it works as you expect. I'll merge the change if so

karan-narula, 2019-09-30:
Hi, thank you for your reply. I have verified in my example that your provided fix works as expected. I can see that in abstract_controller_execution.cpp, if there is a new plan it will check if its empty and change the state accordingly. So you can get away without checking that here.

corot, 2019-10-01:
Good! thanks for trying. PRequested

commit 9023686a (2019-10-02): Update goal pose on replanning, so the feedback remains consistent (#158)

fixes and closes #157
closed 2019-10-02 by spuetz