naturerobots/move_base_flex #161: Robot keeps moving despite cancelling the MoveBaseAction

ChristofDubs, 2019-10-15:
Hello

I have been testing out move_base_flex for a bit, and apart from one issue, it worked pretty well; thanks for all the effort!

The issue that I came across is the following (using Ubuntu 18.04 / ROS melodic):
When using move_base_flex with the ROS planners (global planner, DWA local planner) and commanding it with a SimpleActionClient through the [MoveBaseAction](http://docs.ros.org/kinetic/api/mbf_msgs/html/action/MoveBase.html) interface, calling `action_client_->cancelAllGoals()`  results in the last velocity command to be published; so instead of stopping, the robot keeps moving in a random direction.

I first just added a `publishZeroVelocity();` statement in `AbstractControllerExecution::cancel()` for the case of `if(!controller_->cancel())`, but that only worked sometimes because sometimes the `nav_core_plugin_->computeVelocityCommands(cmd_vel.twist)` in `WrapperLocalPlanner::computeVelocityCommands(...)` seemed to finish later, overwriting the zero velocity command. 

So the way I fixed it is by adding a boolean `cancelled_` flag in the `WrapperLocalPlanner` class that is set to `true` in `WrapperLocalPlanner::cancel()` and to `false` in `WrapperLocalPlanner::setPlan(...)`, and then setting the velocity to 0 after `nav_core_plugin_->computeVelocityCommands(cmd_vel.twist)` for the case of `if (cancelled_)`.

Should I make a PR with this fix? Is there a way to generally prevent the robot from moving in case of cancelling for any planner doesn't support cancelling?

corot, 2019-10-15:
Hi, thanks for reporting. Indeed, makes sense to stop the robot on canceling. Please try new branch https://github.com/magazino/move_base_flex/tree/issue_161. I just added a publishZeroVelocity on breaking the controller loop on canceling, so the zero velocity should not get overwritten.

ChristofDubs, 2019-10-17:
Thank you @corot for your prompt response.

More specifically, I came across two slightly different faulty behaviors during my testing using the released ros-package:
1) a single non-zero setpoint is published after cancelling (happened fairly often)
2) multiple non-zero setpoints are published after cancelling (happens rarely)

I tested with your branch, and it seems to fix 1).

Regarding 2), should I make a new issue or just post more information here? To summarize, It basically seems that `exe_path` is sometimes not actually cancelled (resulting again in the robot keeping to move after cancelling the action).

corot, 2019-10-17:
Can you reproduce 2) ?  Or just happen spuriously? Sounds to me like a different issue

corot, 2019-10-23:
Ok, I have merged the fix and so close the issue. Feel free to open a new one if you can confirm the weird behavior described in 2)
Thanks!

closed 2019-10-23 by corot