naturerobots/move_base_flex #163: cancelling MoveBase Action sometimes doesn't cancel the ExePath

ChristofDubs, 2019-11-01:
Continuation of [Issue161](https://github.com/magazino/move_base_flex/issues/161):

Usually, when cancelling the MoveBase Action, the output looks like this:
```
[AbstractNavigationServer::cancelActionMoveBase]: Cancel action "move_base" 
[AbstractNavigationServer::cancelActionExePath]: Cancel action "exe_path" 
[AbstractControllerExecution::cancel]: Cancel controlling failed or is not supported by the plugin. 
[ControllerAction::run]: Action "exe_path" canceled
```

or, if still in the planning stage:
```
[AbstractNavigationServer::cancelActionMoveBase]: Cancel action
[AbstractNavigationServer::cancelActionExePath]: Cancel action "exe_path"
[AbstractControllerExecution::cancel]: Cancel controlling failed or is not supported by the plugin. Wait until the current control cycle finished!
[AbstractNavigationServer::cancelActionGetPath]: Cancel action "get_path"
[AbstractPlannerExecution::cancel]: Cancel planning failed or is not supported by the plugin. Wait until the current planning finished!
[AbstractPlannerExecution::run]: The global planner has been canceled!
[ControllerAction::run]: Action "exe_path" canceled
```

In some rare cases however, the output just says:
`[AbstractPlannerExecution::cancel]: Cancel planning failed or is not supported by the plugin. Wait until the current planning finished!`

As additional information, right before that 3 MoveBase Actions (frequency: 30Hz) were received and the output looked like:
```
[AbstractNavigationServer::callActionMoveBase]: Start action "move_base"
[AbstractNavigationServer::callActionMoveBase]: Start action "move_base"
[AbstractNavigationServer::callActionMoveBase]: Start action "move_base"
[AbstractNavigationServer::callActionGetPath]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
[AbstractNavigationServer::callActionGetPath]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
[AbstractNavigationServer::callActionGetPath]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
[AbstractPlannerExecution::cancel]: Cancel planning failed or is not supported by the plugin. Wait until the current planning finished!
```
and after that, the robot just kept going for a long time in a non-proper fashion (velocity commands kept changing initially since the local planner still seemed to be executed, but at some point the velocity command froze to some non-zero value). I had added a print at line 340 of `mbf_abstract_nav/src/abstract_controller_execution.cpp` that showed that the `cancel_` flag was `false` when this behavior happened.

How similar is the MoveBase Action code path compared to separate `GetPath` and `ExePath` Actions? I.e. is this issue likely to disappear if I use `GetPath` and `ExePath` Actions instead (as it seems to be the intended usage according to the API)?

corot, 2019-11-04:
Hi, those rare cases are perfectly normal; it just happened that the cancel signal came while the planer was running (we call it at the time at 1 Hz or so). But the behavior shouldn't change.

For the question, MoveBase action is implemented by internally calling the other actions, GetPath, ExePath and Recovery.

I have tried to reproduce the issue with this simple script, without success:

```
#!/usr/bin/env python

import random

import rospy
import actionlib

from geometry_msgs.msg import PoseStamped

from mag_common_py_libs.geometry import quaternion_from_yaw, pose2str

from mbf_msgs.msg import MoveBaseAction, MoveBaseGoal, MoveBaseResult

__author__ = 'santos'

""" 
Call MBF's move_base action with goals provided through RViz on 2D Nav Goal tool, and cancel after a short while.
"""

def create_geo_pose_stamped(x, y, yaw, frame_id='map'):
    pose = PoseStamped()
    pose.pose.position.x = x
    pose.pose.position.y = y
    pose.pose.orientation = quaternion_from_yaw(yaw)
    pose.header.frame_id = frame_id
    return pose

def nav_goal_cb(msg):
    goal = MoveBaseGoal(target_pose=msg)
    move_base_ac.send_goal(goal, done_cb=move_done_cb)
    rospy.loginfo("Calling MBF's move_base action with target pose %s", pose2str(msg))
    rospy.sleep(random.randrange(300, 800)/100.0)
    rospy.loginfo("CANCEL!!!")
    move_base_ac.cancel_goal()

def move_done_cb(status, result):
    if result.outcome == MoveBaseResult.SUCCESS:
        rospy.loginfo("Follow path action succeeded")
    else:
        rospy.logerr("Follow path action failed with error code [%d]: %s", result.outcome, result.message)

if __name__ == '__main__':
    rospy.init_node("call_move_base_action")

    move_base_ac = actionlib.SimpleActionClient("/move_base_flex/move_base", MoveBaseAction)
    move_base_ac.wait_for_server(rospy.Duration(5))

    rospy.Subscriber('/move_base_simple/goal', PoseStamped, nav_goal_cb)

    rospy.spin()
```

Any idea how else can I try to reproduce it?

ChristofDubs, 2019-11-18:
Hi, sorry for the delayed response.

I can reproduce the issue if I change the sleep to `10 ms`; the code in the callback I used is
```
def nav_goal_cb(msg):
    goal = mbf_msgs.MoveBaseGoal(target_pose=msg)
    move_base_ac.send_goal(goal)
    rospy.loginfo("Calling MBF's move_base action with target pose %f %f", msg.pose.position.x, msg.pose.position.y)
    rospy.sleep(0.010)
    rospy.loginfo("CANCEL!!!")
    move_base_ac.cancel_goal()
```
and the output when it failed (i.e. robot just went to the goal anyway) was:
```
move_base_legacy_relay: [nav_goal_cb]: Calling MBF's move_base action with target pose 3.995125 6.590005
move_base: [AbstractNavigationServer::callActionMoveBase]: Start action "move_base"
move_base: [AbstractNavigationServer::callActionGetPath]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
move_base_legacy_relay: [nav_goal_cb]: CANCEL!!!
move_base: [AbstractNavigationServer::cancelActionMoveBase]: Cancel action "move_base"
move_base: [AbstractNavigationServer::cancelActionGetPath]: Cancel action "get_path"
move_base: [AbstractPlannerExecution::cancel]: Cancel planning failed or is not supported by the plugin. Wait until the current planning finished!
move_base: [AbstractNavigationServer::callActionExePath]: Start action "exe_path" using controller "DWAPlannerROS" of type "dwa_local_planner/DWAPlannerROS"
move_base: [DWAPlannerROS::setPlan]: Got new plan
```
It happens about once in 10 tries, but I guess reducing the sleep time would increase the frequency of occurrence. I am not sure though whether that is the same issue as I initially reported, because the output looks pretty different (in the issue in this comment, it just seems `exe_path` is started after receiving the cancel request, so it is executed anyway).

For reference, in the case where the cancelling works correctly, the output I am seeing is
```
move_base_legacy_relay: [nav_goal_cb]: Calling MBF's move_base action with target pose 0.818876 3.458276
move_base: [AbstractNavigationServer::callActionMoveBase]: Start action "move_base"
move_base: [AbstractNavigationServer::callActionGetPath]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
move_base_legacy_relay: [nav_goal_cb]: CANCEL!!!
move_base: [AbstractNavigationServer::cancelActionMoveBase]: Cancel action "move_base"
move_base: [AbstractNavigationServer::callActionExePath]: Start action "exe_path" using controller "DWAPlannerROS" of type "dwa_local_planner/DWAPlannerROS"
move_base: [AbstractNavigationServer::cancelActionExePath]: Cancel action "exe_path"
move_base: [AbstractControllerExecution::cancel]: Cancel controlling failed or is not supported by the plugin. Wait until the current control cycle finished!
move_base: [ControllerAction::run]: Action "exe_path" canceled
```
Thank you for your help so far, please let me know if you need more information.

corot, 2019-11-19:
The difference seems to be that in the first case, the planner action
(get_path) is not yet done when you cancel move_base action.
This seems to me like a bug, as we should handle this better. I'll try to
debug it.
Thanks for reporting in such detail level!

corot, 2019-11-19:
Hello back; yes, I can reproduce the issue. Tentative fix on https://github.com/magazino/move_base_flex/tree/issue_163. Could you give it a fast try? I added ugly log error messages when the change is working (I'll remove before PRequesting this, don't worry)

ChristofDubs, 2019-12-02:
Hi, I got around to test it again.

The fix seems to work, I saw both the `CTRL   DONE` and `PLAN   DONE` output in the log and the robot never kept going.

I have not managed to produce any other failures using that python script; I however encountered some other problems when not using the script: 

1) At some point, the robot kept moving a little bit, then stopped. After this, it didn't respond to any command anymore; both the `/move_base/move_base/result` and `/move_base/move_base/feedback` didn't contain any messages anymore. The log output was:
```move_base: [AbstractNavigationServer::callActionGetPath]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
move_base: [AbstractPlannerExecution::cancel]: Cancel planning failed or is not supported by the plugin. Wait until the current planning finished!
move_base: [ControllerAction::run]: The controller is oscillating for 10.02s
move_base: [AbstractExecutionBase::stop]: Trying to stop the planning rigorously by interrupting the thread!
move_base: [MoveBaseAction::actionExePathDone]: CTRL   DONE   action_state_ = CANCELED   -->   salir
```

To answer my own previous question, the move base action uses its own state machine implementation to call the getPath and exePath action, so these issues generally aren't to be expected when using these actions directly. I have also tried out calling those actions separately and didn't run into any of these problems.

2) I also got a crash:
`move_base: terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
move_base:   what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
move_base: move_base died from signal 6`

of which that stacktrace was:

```#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007f3c865c4801 in __GI_abort () at abort.c:79
#2  0x00007f3c86c19957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f3c86c1fab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f3c86c1eb19 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f3c86c1f488 in __gxx_personality_v0 () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f3c86985613 in ?? () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#7  0x00007f3c86985e95 in _Unwind_Resume () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#8  0x00007f3c83dfaa63 in boost::thread::native_handle() () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#9  0x00007f3c85e03ace in boost::thread::join() () from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#10 0x00007f3c85e15129 in mbf_abstract_nav::AbstractAction<mbf_msgs::GetPathAction_<std::allocator<void> >, mbf_abstract_nav::AbstractPlannerExecution>::start(actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > >&, boost::shared_ptr<mbf_abstract_nav::AbstractPlannerExecution>) () from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#11 0x00007f3c85e758d8 in mbf_abstract_nav::AbstractNavigationServer::callActionGetPath(actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > >) ()
   from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#12 0x00007f3c85e95590 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, mbf_abstract_nav::AbstractNavigationServer, actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > > >, boost::_bi::list2<boost::_bi::value<mbf_abstract_nav::AbstractNavigationServer*>, boost::arg<1> > >, void, actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > > >::invoke(boost::detail::function::function_buffer&, actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > >) ()
   from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#13 0x00007f3c85e956b0 in boost::function1<void, actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > > >::operator()(actionlib::ServerGoalHandle<mbf_msgs::GetPathAction_<std::allocator<void> > >) const () from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#14 0x00007f3c85e9c75f in actionlib::ActionServerBase<mbf_msgs::GetPathAction_<std::allocator<void> > >::goalCallback(boost::shared_ptr<mbf_msgs::GetPathActionGoal_<std::allocator<void> > const> const&)
    () from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#15 0x00007f3c85e8505e in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<mbf_msgs::GetPathActionGoal_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<mbf_msgs::GetPathActionGoal_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<mbf_msgs::GetPathActionGoal_<std::allocator<void> > const>) ()
   from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#16 0x00007f3c85eafe3f in ros::SubscriptionCallbackHelperT<boost::shared_ptr<mbf_msgs::GetPathActionGoal_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
   from /home/christof/ws/devel/lib/libmbf_abstract_server.so
#17 0x00007f3c876b7792 in ros::SubscriptionQueue::call() () from /opt/ros/melodic/lib/libroscpp.so
#18 0x00007f3c87661e8c in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/melodic/lib/libroscpp.so
[... cut ...]
    rospy.loginfo("CANCEL!!!")
    move_base_ac.cancel_goal()
```

I primarily reported these issues because it would be wrong to claim that the fix fixed all problems that I encountered, so in my mind, reporting those issues is the correct thing to do. I however spent quite some time now reproducing these issues using the Move Base Action interface, which I ultimately would not recommend to use anyway, since using the getPath and exePath actions directly is much more powerful.

Thanks anyway and keep up the good work.

corot, 2019-12-02:
Hi, thanks for the extensive testing. So you think the fix is worth merging, regardless this other problem you have encountered?
Actually, if you can reproduce it, even only part of the time, I invite you to open another issue, so I can close this one with the actual fix.

ChristofDubs, 2019-12-04:
Hi, it worked better with the fix for me, but I cannot judge whether this fix would introduce potentially unwanted side effects, so you'd have to take that call yourself. I for example didn't test any case where I cancelled the action during recovery.

I have not been able to reproduce the other issue again; I will open a new issue if I do.

corot, 2020-02-04:
Seems to work in all cases, so closing

closed 2020-02-04 by corot