naturerobots/move_base_flex #165: controller_frequency automatically resets to 20 (default value)

SelvamArul, 2019-12-06:
I am using the `move_base_flex` on **ros melodic**
I noticed the `controller_frequency` parameter automatically resets to 20 (default value provided [here](https://github.com/magazino/move_base_flex/blob/6a069cf8d22bdc4493627c2fc71eb4cecb29cacc/mbf_abstract_nav/src/mbf_abstract_nav/__init__.py#L25)) despite setting it to a different value during launch. 
I tested if the param is set successfully during launch and launch does set the param successfully. But then it is automatically set to 20 by some other process. Could this be a possible bug?

corot, 2019-12-08:
Can you post your launch file? I suspect what it's happening here. If you are launching the `move_base_legacy_relay.py` script together with  `mbf_costmap_nav` node, the former node's configuration will overwrite the later's one. You check this by opening the dynamic reconfigure tool and check if there are a move_base node with a configuration similar to move_base_flex.

SelvamArul, 2019-12-09:
Yes, I am launching  `move_base_legacy_relay.py`script together with  `mbf_costmap_nav` node.
My launch file looks like:

    <?xml version="1.0"?>
    <launch>

        <node pkg="mbf_costmap_nav" type="mbf_costmap_nav" respawn="false" name="move_base_flex" output="screen">

            <param name="controller_frequency"  type="double" value="10.0" />

            <rosparam file="$(find mario_launch)/navigation/planners.yaml" command="load" />
            <rosparam file="$(find mario_launch)/navigation/mbf_controllers.yaml" command="load" />

            <rosparam file="$(find mario_launch)/navigation/costmap_common_params.yaml" command="load" />
            <rosparam file="$(find mario_launch)/navigation/local_costmap_params.yaml" command="load" />
            <rosparam file="$(find mario_launch)/navigation/global_costmap_params.yaml" command="load" />
            <remap from="/cmd_vel" to="/mario_kart/velocity" />
        </node>
        <node pkg="mbf_costmap_nav" type="move_base_legacy_relay.py" name="move_base_legacy_relay" />
    </launch>

But, `dynamic_reconfigure` does not list any move_base related params. Here is the whole list of params listed:

> /amcl
> /arm_cam
> /arm_cam/image_raw/compressed
> /arm_cam/image_raw/compressedDepth
> /arm_cam/image_raw/theora
> /gazebo
> /mario/arm_controller/gains/arm_elbow_joint
> /mario/arm_controller/gains/arm_shoulder_lift_joint
> /mario/arm_controller/gains/arm_shoulder_pan_joint
> /mario/arm_controller/gains/arm_wrist_1_joint
> /mario/arm_controller/gains/arm_wrist_2_joint
> /mario/arm_controller/gains/arm_wrist_3_joint
> /mario/back_left_pan_joint_position_controller/pid
> /mario/back_right_pan_joint_position_controller/pid
> /mario/front_left_pan_joint_position_controller/pid
> /mario/front_right_pan_joint_position_controller/pid
> /move_base_flex
> /move_base_flex/dwa
> /move_base_flex/global_costmap
> /move_base_flex/global_costmap/inflator
> /move_base_flex/global_costmap/static_layer
> /move_base_flex/global_planner
> /move_base_flex/local_costmap
> /move_base_flex/local_costmap/inflator
> /move_base_flex/local_costmap/obstacles
> /move_base_flex/local_costmap/static_layer
> /move_base_legacy_relay

corot, 2019-12-09:
yes, yes; you are not loading move_base legacy relay params, but still,
move_base dynamic reconfigure will be there, and by default will set 20 Hz
for controller freq.
You can try to
A) do not launch move_base_legacy_relay
B) load your desired controller frequency within move_base_legacy_relay,
something like:

set param /move_base_legacy_relay/controller_frequency XXX

SelvamArul, 2019-12-17:
@corot Thanks for the reply. It makes sense.

closed 2019-12-17 by SelvamArul

Ngochuy2137, 2022-09-16:
@corot hello, can you tell me why we need to run move_base_legacy_relay node ? I don't understand the function of node at all!

corot, 2022-09-17:
> @corot hello, can you tell me why we need to run move_base_legacy_relay node ? I don't understand the function of node at all!

It's only needed if you want to use the old move_base interface defined on nav_msgs, or the simple goal sent by rviz.
That is, it's not mandatory