naturerobots/move_base_flex #167: controller updates in "real"-time in simulation

MCFurry, 2020-01-08:
While running mbf in a simulation where the simulation doesn't reach (or is not set to) a 1.0 real-time factor, mbf still calls controller execution in "real"-time since it uses the boost chrono timer:
[link](https://github.com/magazino/move_base_flex/blob/e4cdbb9f9ca0191cfd1905ff5125f67685292534/mbf_abstract_nav/src/abstract_controller_execution.cpp#L375)
This way the performance of the controller largely depends on how fast we (can) run a simulation and thus yields different results, or even problems in some scenarios, since the ROS time is not synchronous with the execution time of the controller.

Is this intended behaviour? Or how would you advise to run simulations with mbf?

spuetz, 2020-01-08:
You're right! Thank you for mentioning this error. Honestly, I didn't think about it.

MCFurry, 2020-01-09:
So maybe switching the timing to ROS-based timing would be the solution?
Would this in your opinion be "the best" solution?

corot, 2020-02-03:
I suppose using [ROS Timers](http://wiki.ros.org/roscpp/Overview/Timers) will do the trick.
PR welcome!!!

corot, 2021-01-29:
fixed on master

closed 2021-01-29 by corot