naturerobots/move_base_flex #176: Turtlebot3 mbf example

Rayman, 2020-03-05:
Hey,

I've made a package with launch files for running mbf on a turtlebot3:
https://github.com/Rayman/turtlebot3_mbf

Do you think these are good enough to include in the documentation? I think it will give some people a good starting point for moving to this package.

corot, 2020-03-10:
Sure, would make a nice tutorial. I gave a fast try and got stuck in:

```
$ roslaunch turtlebot3_mbf amcl_demo.launch 
....
....
process[robot_state_publisher-1]: started with pid [11562]
process[map_server-2]: started with pid [11563]
process[amcl-3]: started with pid [11568]
process[move_base_flex-4]: started with pid [11581]
process[move_base_legacy_relay-5]: started with pid [11586]
[/amcl  WARN 1583855480.825892704]: Request for map failed; trying again...
```

I'll try to find time to debug what's going wrong

simonbogh, 2020-10-24:
@Rayman I think this is a great idea, thank you. I didn't find any other tutorials or documentation giving a minimal working example replacing move_base (maybe I missed it somewhere).

corot, 2020-10-26:
> @Rayman I think this is a great idea, thank you. I didn't find any other tutorials or documentation giving a minimal working example replacing move_base (maybe I missed it somewhere).

Right, right;  you are absolutely right. I'll add a link to @Rayman's TB3 bringup to the wiki.

corot, 2020-11-06:
Added to Getting Started section of the [wiki](wiki.ros.org/move_base_flex)

closed 2020-11-06 by corot