naturerobots/move_base_flex #177: CostmapController plugin problem

q576333, 2020-03-17:
Hi, First of all. Thanks for your good job.
Now, I want to write my own local planner so I see the move_base_flex tutorial page.
It says you have two choice, one is inherit from nav_core::BaseLocalPlanner another is mbf_costmap_core::CostmapController.
mbf_costmap_core::CostmapController plugin is made by yourself, so you have provide one example (your_planner.h) in tutorial page, also I just need implement three function.
But I try to trace source code, you are define the five pure virtual function in costmap_controller.h, so Is the ros tutorial page is wrong? 
ros tutorial page:
http://wiki.ros.org/move_base_flex/Tutorials/NavCoreBasedLocalPlannersToMBF

And the second problem. In the README mention this package can easily implement in the different map representation to do global planning and local planning but in the CostmapController and CostmapPlanner plugin also need pass the costmap_2d class pointer to initialize, so do I need define my own global and local planner plugin? or Is there any way to use the plugin provided by this package?

Rayman, 2020-03-17:
Yes, you have the implement those 5 functions. I think the guide is a bit outdated. For `cancel` you can return false to have the default behavior.

I dont understand your second question. Do you want to make a local planner without a costmap_2d?

q576333, 2020-03-17:
Yes, I want implement global and local planning with different map representation in the future, so is there any suggestion?

spuetz, 2020-03-17:
What kind of map representation do you want to use? This is a bit more work than just writing a planner and controller for costmap_2d, but it depends on what you want to do in detail.

q576333, 2020-03-18:
So far, I found one good work is from ethz asl lab, It's using 3D reconstruction technique to representation the environment, called voxblox. Therefore, I think if I use this map representation to do global and local planning, There is two choice to do that, first is define new planner plugin to assign map class to support planning. Another way is continue using original plugin and load costmap_2d class, but didn't use that. Instead, create own class and initialize in orginal plugin function. Which is better choice?

In addition, I saw you have the similar work that use 3D information to do mobile robot motion planning, Do you have any suggestion or reminder in this topic? Because I am doing research on that. I hope I can contribution this package in the future to develop using 3D information to do mobile robot motion planning.

Thanks for your reply.

spuetz, 2020-06-12:
We are currently planning an mbf_voxblox_server. If you started work on this already, let me know.

spuetz, 2020-06-12:
I updated the Tutorial

closed 2020-06-12 by spuetz