naturerobots/move_base_flex #178: Strange MBF crash when requesting a path close to costmap border

doisyg, 2020-03-25:
Hello,
I noticed a crash happening in strange conditions:

- When no global plan (using base local planner) has been generated before, and,
- When requesting a path with the `get_path` action with an end point close to the border of the global costmap (for instance x=0.0 and y=0.0)

Steps to reproduce:

1.  Start `mbf_costmap_nav`
2. Use `get_path` action with, for instance:
```
use_start_pose: False
start_pose: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  pose: 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 0.0
target_pose: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: 'map'
  pose: 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
tolerance: 0.0
planner: ''
concurrency_slot: 0
```

What's really strange is that if the `get_path` action is used before with a valid destination (a pose in the middle of the global costmap with no surrounding obstacles for instance), a new call to the `get_path` action with x=0.0 and y=0.0 works: it responds properly that there is not path and MBF is not crashing. I suspect it to be linked to calls to the global planner...

Side notes: the `check_pose_cost` service seems to be also affected by a different behavior before and after a first call to the global planner when requesting a check at x=0.0 and y=0.0 on the global costmap.

corot, 2020-03-26:
Thanks for reporting; which global planner are u using?

doisyg, 2020-03-26:
I edited the issue, the global planner used is `"GlobalPlanner", type: "global_planner/GlobalPlanner"`
My current ugly fix is to call `get_path`,  just after the start of MBF, with a pose that i know to be free as the start and target pose (which forces an initial call to the global planner)

corot, 2020-03-26:
I get the expected "Planner failed". So,,,

- Which ROS are you using?
- Are u compiling MBF and/or navigation?  or using .debs?

doisyg, 2020-03-26:
Melodic, Ubuntu 18.04 and last debs.
I ll try to set up a repo with the config to reproduce it if you cannot

corot, 2020-03-26:
Can u try also a different global planner, like Navfn?

doisyg, 2020-03-26:
Same config but just using `navfn/NavfnROS` instead, no issue:
```
[ INFO] [/move_base]: Start action "get_path" using planner "GlobalPlanner" of type "navfn/NavfnROS"
[ INFO] [/move_base]: Planning reached max retries! (0)
```

Whereas with `global_planner/GlobalPlanner`:
```
[ INFO] [/move_base]: Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"
[move_base-6] process has died [pid 30652, exit code -11, cmd /opt/ros/melodic/lib/mbf_costmap_nav/mbf_costmap_nav /cmd_vel:=/twist_mux_root/twist_nav __name:=move_base __log:=/home/gd/.ros/log/73528660-6f7f-11ea-9009-f0d5bfb298e1/move_base-6.log].
log file: /home/gd/.ros/log/73528660-6f7f-11ea-9009-f0d5bfb298e1/move_base-6*.log
```
ps: the name of the mbf node is move_base for legacy parameter name conservation, but it is mbf_costmap_nav which is running

doisyg, 2020-03-26:
Reproduced in this repo: https://github.com/wyca-robotics/mbf_test

1. Clone in workspace
2. `roslaunch teb_local_planner_tutorials robot_diff_drive_mbf_in_stage.launch`
3. Call the `get_path` action with a target_pose of (0,0,0) and it crashes. To call the action, I use `rosrun actionlib axclient.py /move_base/get_path` with:
```
use_start_pose: False
start_pose: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  pose: 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 0.0
target_pose: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: 'map'
  pose: 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
tolerance: 0.0
planner: ''
concurrency_slot: 0
```
 
I had to remove the border of the map as the issue seems to happen only with empty cells. Also I suspected the outlineMap parameter recently merged, see https://github.com/ros-planning/navigation/pull/926,  to play a role (as the costmap is not outlined at initialization), so I disabled it ( `<param name="GlobalPlanner/outline_map" value="false" />`). But the crash happens also without the outline.

corot, 2020-03-27:
I wonder if it can be related to [this issue](https://github.com/ros-planning/navigation/pull/897).... Most  probably not; global_planner changes the underlying map itself, but unless this has changed recently, only on robot position.

doisyg, 2020-03-27:
It is indeed not related to MBF (but MBF definitely helped the diagnostic), I just replicated it with legacy move base. I will open an issue on navigation.
Should I close this one?

Rayman, 2020-03-27:
@doisyg NavFn was the predecessor to global_planner. It is battle tested and works fine.

corot, 2020-03-30:
Closed as out of scope

closed 2020-03-30 by corot