naturerobots/move_base_flex #181: mbf_tolerance_check not dynamically reconfigurable

mhoellmann, 2020-04-08:
It would be nice if the mbf_tolerance check would be dynamically reconfigurable so that the tolerance check can be adjusted for different types of target poses.

The parameters of interest are mbf_tolerance_check, dist_tolerance and angle_tolerance:

https://github.com/magazino/move_base_flex/blob/c16d21b1a7364c960aa1bbae36221f4b7e77481d/mbf_abstract_nav/src/abstract_controller_execution.cpp#L69-L71

corot, 2020-04-09:
Sounds reasonable. But by other hand, all local planners I can think about have their own dynamically reconfigurable tolerance parameters, so those ones look redundant (and confusing, given the duplication). Moreover, those parameters cannot overrule controller's ones: look at this line: https://github.com/magazino/move_base_flex/blob/c16d21b1a7364c960aa1bbae36221f4b7e77481d/mbf_abstract_nav/src/abstract_controller_execution.cpp#L243. There's an `OR`. To make these parameters really useful, we shouldn't ask the controller if `mbf_tolerance_check` is true.

So I propose:
- remove these parameters
- add them to the action goal (same as planner)
- change the above line so this parameters overrule controller's ones
- modify planner action to have both dist and angle tolerances

Sounds reasonable? @spuetz, your thoughts?

spuetz, 2020-04-09:
What? The OR makes sense, there are two more lines ;) It means that either the controller plugin returns true, if the goal is reached with respect to the tolerance values or mbf says so, if activated. I suggested @mhoellmann to open this issue, since we have this issue with one of our robots, I will implement something solving this the next days. The `mbf_tolerance_check` param enables mbf to check for reached goal. This is necessary since many controllers have not implemented the tolerance check in `isGoalReached`. 
This parameter makes absolutely sense. I propose: 
- Adding the `angle_tolerance` and the `dist_tolerance` parameter to the action goal in `ExePath.action`.
- Overwriting the parameters in the corresponding controller execution -> implement this functionality in the `AbstractControllerExecution` class.

corot, 2020-04-09:
But,,,  let's say u want more precision for a particular goal; with current code you cannot overrule plugin tolerances with lower values! Moreover, having one tolerance or the other, not both, is much more predictable and intuitive, imho.

Also, adding the tolerances to the goal makes the params redundant, isn't it? Having 3 places to tinker with tolerance is gonna be very confusing to users.

spuetz, 2020-04-09:
The values you specify in the action should be the values handed over to the plugin and also to the mbf tolerance check. If mbf should check the arrival can be set by the param.
Further, we can not simply remove a param, due to compatibility (first we would need to make it deprecated), and I think we also should not remove it. I use it often, since many old (wrapped) plugins do not support this functionality.
Adding the tolerance (default -1) let you specify specific values for certain plugins, whereas the global param works for all in the first place.
Software can be confusing if it is not well documented. So yes we would need to document this feature well.

corot, 2020-04-10:
But you cannot pass the values to the plug-ins,,, there's no interface for
that. But we can use action goal values to overwrite mbf ones, and overrule
plug in ones

I'm good with keeping the mbf parameters, but I would make them overrule
(as goal ones) plug in parameters

commit 134ea6eb (2020-04-17): mbf_tolerance check for missions (#182)

* add goal_tolerance to action msg
* set tolerance of the mission through setNewPlan method
* closes #181
closed 2020-04-17 by spuetz