naturerobots/move_base_flex #183: extensive get_path requests makes move_base_flex stop responding

ChristofDubs, 2020-04-21:
In rare cases, issuing a new `get_path` request immediately after receiving the result of the previous one results in a `Start action "get_path" using planner "GlobalPlanner" of type "global_planner/GlobalPlanner"` confirmation, immediately (a few milliseconds) followed by a `Cancel planning failed or is not supported by the plugin. Wait until the current planning finished!` warning, and rhen no more messages being published on `get_path/result`.

To reproduce this, a goal can be sent
```
void sendRequest() {
	ac->sendGoal(goal,
		[](const actionlib::SimpleClientGoalState& state, const mbf_msgs::GetPathResultConstPtr& result) {
			onResult(result);
		},
		[]() {}, [](const mbf_msgs::GetPathFeedbackConstPtr& feedback) {});
}
```
with a callback that re-sends the goal:
```
void onResult(const mbf_msgs::GetPathResultConstPtr& result) {
	if (!result) {
		ROS_WARN("Empty result.");
		return;
	}  
	sendRequest();
}
```

It took usually around 5 minutes of constant `get_path` requests until the move_base_flex became unresponsive; in parallel ran `sudo stress --cpu 4 --io 3 --vm 2 --vm-bytes 256M` to add some additional CPU load, but that may not be necessary to reproduce it.

spuetz, 2020-04-21:
All action goals on the same concurrency slot?

ChristofDubs, 2020-04-21:
> All action goals on the same concurrency slot?

Yes.

@corot already put together [this branch](https://github.com/magazino/move_base_flex/tree/get_path_freeze) that seems to fix it

corot, 2020-04-22:
fixed on master. @ChristofDubs, can you confirm, please?

closed 2020-04-22 by corot

ChristofDubs, 2020-04-23:
> fixed on master. @ChristofDubs, can you confirm, please?

The `get_path` behavior seems fine now. However, I am facing some more severe problem now:
When I send 2D nav goals (uses MoveBaseAction), it works fine as long as i send a new goal before the robot arrived. If I send a goal after the robot arrived, the robot doesn't move, and after starting the action, the following error appears (6ms later):
```
[ros.mbf_abstract_nav.exe_path]: Start action "exe_path" using controller "DWAPlannerROS" of type "dwa_local_planner/DWAPlannerROS"
[ros.mbf_abstract_nav.actionlib]: To transition to a cancelled state, the goal must be in a pending, recalling, active, or preempting state, it is currently in state: 3
```
The problem doesn't seem to be related with the MoveBaseAction per se; this is just the easiest way to reproduce the problem.

ChristofDubs, 2020-04-23:
Above problem was fixed by [ee55488](https://github.com/magazino/move_base_flex/commit/ee554888d6014f2f5d1a27ffd341c0d7fb799557) and I have not encountered any other problem.

Once more, thanks for your help and fast response to the issue.